bigtreetech/SKR-2 #112: BTT SKR 2 heating HE1 Problem

GrimmaPL, 2022-01-27:
Hello everyone. 
I have a strange problem.  For not knowing reasons one of my two hot end's stop working.  But I will start from the beginning. 
Few days ago I just finished upgrading my old Wanhao i3.
I have installed BTT SKR 2 with  tmc 2209 drivers, X3D dual nozzle himera hot end, 3Dtouch, new power supply, external mosfet for heated bed,  extra fans for better motherboard temperature control. I have changed almost everything except the frame and stepper motors. 
Marlin 2.0 was little bit tricky,  but finally everything was working quite nicely. 
I was able to do few shorter and longer prints without problem.
Until yesterday...
Like I said before, one of my heating cartridges don't want to cooperate anymore. 
I've already checked thermistor,  resistance on cartridge, voltage on power supply and on the pins HE1 and wiring. Everything is like it should be. 
I even changed the thermistor and cartridge for new one, just in case.  
I'm out of ideas what can be the reason of this issue, and I'm hoping that maybe you guys can't help me to find it. 

Ps. I'm aware that my English isn't perfect and I hope that you read and understand everything without a stroke 😉

radek8, 2022-01-27:
If it doesn't melt. everything can't be okay.
1. What is the resistance of the heater
2. Switch on the heating and measure the voltage on the radiator.
3. Switch off the heating and measure the voltage on the heater.

GrimmaPL, 2022-01-27:
> If it doesn't warm up, everything can't be okay.
> 
> 1. What is the resistance of the heater
> 2. Switch on the heating and measure the voltage on the heater.
> 3. Switch off the heating and measure the voltage on the heater.

A.1 resistance is 3.7 ohm
A.2. Voltage when heating is ON 12.13V 
A.3.  Voltage when heating is OFF -0.243V

Second and third was checked with heating cartridge dismounted.

radek8, 2022-01-27:
Perform the measurement with the heater connected

radek8, 2022-01-27:
You should not measure negative voltage.
Do you measure the voltage at the right points?