bigtreetech/SKR-2 #114: Tft35 e3 v3.0 "no printer attached"

Wakanapi, 2022-01-27:
Hey, I can't connect my printer with E3 v3 screen. I have changed ports several times, and also the baud rate, from marlin and from the Tft menu.

mslucass, 2022-01-28:
If you go to Marlin mode on the display can you print? Or is this error only on the touchscreen mode? 
I had this error only on touchscreen mode and it took me days but in the end I had a loose cable connection between de SKR-2 and the touchscreen.
ESP1 and ESP2 for the Marlin mode
and the black 5 pin wire for the touchscreen. This wire was loose in my case

Wakanapi, 2022-01-28:
Yes, only in Touchsreen mode. I plug the TFT pin into the SWD connector. Thanks for making me recheck the cable connections. Now everything works fine.

closed 2022-01-28 by Wakanapi