bigtreetech/SKR-2 #115: Cannot upload firmware with platformio

MingyaoLiu, 2022-01-29:
Building the firmware works fine, however it cannot upload the firmware to SKR2 (429)
Log:
```
...
...
...
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\BIGTREE_SKR_2_F429\src\src\inc\Warnings.cpp.o
Checking size .pio\build\BIGTREE_SKR_2_F429\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.2% (used 17304 bytes from 131072 bytes)
Flash: [==        ]  20.5% (used 214456 bytes from 1048576 bytes)
Configuring upload protocol...
AVAILABLE: dfu, jlink, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\BIGTREE_SKR_2_F429\firmware.bin
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

embedded:startup.tcl:26: Error: Can't find board/stm32f429.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 26
*** [upload] Error 1
============================================= [FAILED] Took 5.26 seconds =============================================

Environment         Status    Duration
------------------  --------  ------------
BIGTREE_SKR_2_F429  FAILED    00:00:05.264
======================================= 1 failed, 0 succeeded in 00:00:05.264 ======================================= 
The terminal process "C:\Users\mingy\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1.
```

Above in log the error seems to be `embedded:startup.tcl:26: Error: Can't find board/stm32f429.cfg`

Am I on a old version of a driver or something? 

Update the firmware with SD card works fine. Any help would be appreciated.

MingyaoLiu, 2022-01-29:
I think I'm just retarded ? 
STLink uses SWD header not USB right...

Someone confirm this and I will close it

MingyaoLiu, 2022-01-29:
I'm retarde.

closed 2022-01-29 by MingyaoLiu

MarkEhr, 2022-04-14:
Hi @MingyaoLiu I'm having the exact same error while uploading. How did you fix it?

MingyaoLiu, 2022-04-14:
You can't upload to it, you have to build the image, put the image in sd card, then put sd card onto skr2, boot skr2 it will flash the firmware after a few seconds.

If you want to use the upload, you may need to get a ST32 uart programmer cable, but that's risky from what i learnt