bigtreetech/SKR-2 #117: SKR 2 wont start

DinWilliam, 2022-01-30:
Help

Just installed SKR 2 on my ender 5 but after turning it on there is only blue screen on the LED. If i connect my TFT35 it just says "printer not recognised". Constant red led light on the board. Please help.

radek8, 2022-01-31:
Send wiring photos and configuration files.

Longonwoo, 2022-02-01:
when you code your firmware you have to set your com port 1 value to 0 and your baud rate to 11250 then your boards will be able to talk to each other.  It took me a couple of sleepless days to figure it out myself so dont feel bad

Longonwoo, 2022-02-01:
Have you tried booting it in Marlin mode?  It may connect there even if your BTT screen does not

radek8, 2022-02-01:
> when you code your firmware you have to set your com port 1 value to 0 and your baud rate to 11250 then your boards will be able to talk to each other. It took me a couple of sleepless days to figure it out myself so dont feel bad

Attention!
SKR2 uses port 1 for communication with TFT.
In Marlin, therefore, you must define port 1 on 1.
Default TFT serial port speed at 115200 (not 11520).
The serial port can change to 250000 in the configuration.

radek8, 2022-02-01:
Connection instructions are here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

[image]

radek8, 2022-02-01:
Marlin Dependencies
Firmware source: https://github.com/MarlinFirmware/Marlin/releases

Minimum Marlin firmware version: 2.0.8.1

Distribution date: 2021-05-15

In order the TFT firmware is able to provide all of its functionalities/features, the following options must be enabled in Marlin firmware.

General options which MUST be activated:

EEPROM_SETTINGS (in Configuration.h)

BABYSTEPPING (in Configuration_adv.h)

AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)

AUTO_REPORT_POSITION (in Configuration_adv.h)

M115_GEOMETRY_REPORT (in Configuration_adv.h)

M114_DETAIL (in Configuration_adv.h)

REPORT_FAN_CHANGE (in Configuration_adv.h)

Options to support printing from onboard SD:

SDSUPPORT (in Configuration.h)

LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)

AUTO_REPORT_SD_STATUS (in Configuration_adv.h)

SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)

Options to support dialog with host:

EMERGENCY_PARSER (in Configuration_adv.h)

SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)

HOST_ACTION_COMMANDS (in Configuration_adv.h)

HOST_PROMPT_SUPPORT (in Configuration_adv.h)

Options to support M600 with host & (Un)Load menu: (Options to support dialog with host as pre requisite)

NOZZLE_PARK_FEATURE (in Configuration.h)

ADVANCED_PAUSE_FEATURE (in Configuration_adv.h)

PARK_HEAD_ON_PAUSE (in Configuration_adv.h)

FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)

Options to fully support Bed Leveling menu:

Z_MIN_PROBE_REPEATABILITY_TEST (in Configuration.h)

G26_MESH_VALIDATION (in Configuration.h) (in Configuration.h)

Z_STEPPER_AUTO_ALIGN (in Configuration_adv.h)

Longonwoo, 2022-02-01:
Yes I know that's what the documentation states.  But I had the same issue after setting it up that way.  I ended up watching youtube videos to see what others had done to get the BTTtft35 to work with a BTT skr board.  I was able to boot.  I could link to pronterface and get the board to respond and communicate with my system.  I could even boot the BTTtft35 to work in Merlin mode but I got the same error "printer not recognized" when ever I went into the graphical configuration.  

I found https://www.youtube.com/watch?v=57f-sG0nFeM&t=1329s from Chris Riley and he had to set com1 to 0 to get the graphical to communicate and yes I am sorry I left a 0 off my baud rate of 112500.  But once I made those changes to my firmware my system could talk between the two boards.  

Just sharing my experience

I wish you all the best of luck.

radek8, 2022-02-01:
If your TFT is communicating with Marlin on port 0, then you do not have a SKR2 blanket (the ticket writer writes that he has SKR2, and this is the repository for SKR 2), but some other, such as SKR V1.3 or SKR V1.4.
SKR V1.3 (1.4) use port 0.
It's great that you share your experience, but you have to be precise.
Otherwise, you will mislead inexperienced users.

radek8, 2022-02-01:
The serial port speed is not 112500 but 115200!

DinWilliam, 2022-02-01:
Thanks for all reply. Got called to work so wont be home until thursday or friday. Will try everyones responce then.

DinWilliam, 2022-02-04:
[Uploading Help.zip…]()

DinWilliam, 2022-02-04:
[Uploading Help.zip…]()
I cant upload files it seems. I got the marling screen to work but not the touchscreen

DinWilliam, 2022-02-04:
// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_V2_0_REV_B 
#endif

/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 1

/**
 * Serial Port Baud Rate
 * This is the default communication speed for all serial ports.
 * Set the baud rate defaults for additional serial ports below.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 115200
//#define BAUD_RATE_GCODE     // Enable G-code M575 to set the baud rate

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
 * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1
#define BAUDRATE_2 115200   // Enable to override BAUDRATE

/**
 * Select a third serial port on the board to use for communication with the host.
 * Currently only supported for AVR, DUE, LPC1768/9 and STM32/STM32F1
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_3 2
#define BAUDRATE_3 115200   // Enable to override BAUDRATE

// Enable the Bluetooth serial interface on AT90USB devices
//#define BLUETOOTH

// Name displayed in the LCD "Ready" message and Info menu
#define CUSTOM_MACHINE_NAME "Ender-5 Pro"// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_V2_0_REV_B 
#endif

/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 1

/**
 * Serial Port Baud Rate
 * This is the default communication speed for all serial ports.
 * Set the baud rate defaults for additional serial ports below.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 115200
//#define BAUD_RATE_GCODE     // Enable G-code M575 to set the baud rate

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
 * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1
#define BAUDRATE_2 115200   // Enable to override BAUDRATE

/**
 * Select a third serial port on the board to use for communication with the host.
 * Currently only supported for AVR, DUE, LPC1768/9 and STM32/STM32F1
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_3 2
#define BAUDRATE_3 115200   // Enable to override BAUDRATE

// Enable the Bluetooth serial interface on AT90USB devices
//#define BLUETOOTH

// Name displayed in the LCD "Ready" message and Info menu
#define CUSTOM_MACHINE_NAME "Ender-5 Pro"

radek8, 2022-02-04:
> Uploading Help.zip… I cant upload files it seems. I got the marling screen to work but not the touchscreen

You have to wait for the attachment to save, otherwise it will not open ...

radek8, 2022-02-04:
Port 1 is OK, speed 115200.
You must set a parameter in the config.ini file
serial_port: P1: 6 P2: 7 P3: 0 P4: 0
save and conficg.ini save to sd card, insert card into TFT and turn on display.

DinWilliam, 2022-02-04:
[Help.zip](https://github.com/bigtreetech/SKR-2/files/8005177/Help.zip)

radek8, 2022-02-04:
You have a bad TFT cable connection

radek8, 2022-02-04:
[image]

radek8, 2022-02-04:
[image]

DinWilliam, 2022-02-04:
Yes, it works with the new cable connection. .But every guide i checked said to connect to the pins i did or i have done some bad research =S thank you for your help and thanks everyone for your time

radek8, 2022-02-04:
Probably someone posted bad information or made a bad video on Youtube.
You are not the first with this mistake.

radek8, 2022-02-04:
If your problem is solved, close the ticket.

closed 2022-02-04 by DinWilliam