bigtreetech/SKR-2 #118: 3D Touch stopped during "auto home"

valentin-harrang, 2022-02-03:
Hi,

I added a 3D Touch on my `SKR 2 motherboard` on my CR10 and when I do `auto home`, the 3D Touch probes in a vaccum, the Z axis does not come down, the 3D Touch blinks and it says `stopped` on the screen.

I think the 3D Touch is plugged in correctly because I followed the Bigtreetech SKR 2 pin diagram (brown - red - yellow - black - white).

`Configuration.h`: https://pastebin.com/6Gb7hJh6
`Configuration_adv.h`: https://pastebin.com/cmAZFdb2

[image]

[image]

Video: https://user-images.githubusercontent.com/19478137/152341304-31f234f7-2ef7-4ac2-8072-88782862b7e1.mp4

I found someone who has the same problem: 

[https://3dprinting.stackexchange.com/.../ender-3-pro-will...](https://3dprinting.stackexchange.com/questions/15799/ender-3-pro-will-not-auto-home-correctly-after-mainboard-bltouch-upgrades)

Do you know how to fix the problem?

Thank you 🙏

radek8, 2022-02-03:
You have a configuration error.
You must set:
// # define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING

radek8, 2022-02-03:
The first line says that the limit switch on the Z axis is connected to the _Z-stop_ connector
The second line says that the limit switch (probe) is connected to the _Probe_ connector

You enable the line to which the switch (probe) is connected.
Your probe is in the _probe_ connector.

radek8, 2022-02-03:
If you want to use the touch screen that is in the video, enable the option
#define AUTO_REPORT_FANS

valentin-harrang, 2022-02-03:
Thanks for the answers. I managed to fix the problem by connecting the 2 pins (white and black) on the Z-Stop

radek8, 2022-02-04:
This was the second variant of the solution:)
Adjusting Marlin and leaving the connector on the PROBE pins was a better solution (from an aesthetic point of view).
The connector from the probe would be together and the Z-STOP connector could be used for other peripherals next time

radek8, 2022-02-04:
The Probe connector is more suitable for the probe. It has no resistors and a capacitor.

[image]

radek8, 2022-02-20:
Please close the ticket