bigtreetech/SKR-2 #121: What am I missing?

Eddiiie, 2022-02-14:
Platformio  SK2

Marlin from BTT Githib,  Marlin2 bug fix from Marlin GITHIB..

platformio

Tried all sorts of  default envs but always get same error -  Error: Unknown board ID 'marlin_STM32F407VGT6_CCM'

#default_envs = BIGTREE_SKR_2
#default_envs = BIGTREE_SKR_2_F429
default_envs = BIGTREE_SKR_2_USB

spent hours trying to solve and search    does not seem to be easy how to  for this board.

I also have SKR Turbo, it compiles just fine.

closed 2022-02-14 by Eddiiie

binhamed84, 2022-02-14:
The name of board you need to write in configuration.h maybe you mean this board name:

// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD **BOARD_BTT_SKR_V2_0_REV_B**
#endif

------

And the evns is on platformio.ini file most be:
default_envs = BIGTREE_SKR_2_F429

Eddiiie, 2022-02-14:
That too!   But really, nothing would event start compiling until I moved Marlin to the C drive.  I guess paths are too long.
Thank you for the reply.