bigtreetech/SKR-2 #122: M122 Says Enabled  False

Eddiiie, 2022-02-16:
SKR2  TMC2209  Marlin 2

M122 reports False for enabled.   Nothing moves.

What am I missing?  platformio

configuration.h
#define MOTHERBOARD BOARD_BTT_SKR_V2_0_REV_B

#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 100 }
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 }

configuration_adv.h

  #if AXIS_IS_TMC(X)
    #define X_CURRENT       650        // (mA) RMS current. Multiply by 1.414 for peak current.
    #define X_CURRENT_HOME  X_CURRENT  // (mA) RMS current for sensorless homing
    #define X_MICROSTEPS     16        // 0..256
    #define X_RSENSE          0.11
    #define X_CHAIN_POS      -1        // -1..0: Not chained. 1: MCU MOSI connected. 2: Next in chain, ...
    //#define X_INTERPOLATE  true      // Enable to override 'INTERPOLATE' for the X axis
    //#define X_HOLD_MULTIPLIER 0.5    // Enable to override 'HOLD_MULTIPLIER' for the X axis
  #endif

  #if AXIS_IS_TMC(X2)
    #define X2_CURRENT      800
    #define X2_CURRENT_HOME X2_CURRENT
    #define X2_MICROSTEPS    X_MICROSTEPS
    #define X2_RSENSE         0.11
    #define X2_CHAIN_POS     -1
    //#define X2_INTERPOLATE true
    //#define X2_HOLD_MULTIPLIER 0.5
  #endif

  #if AXIS_IS_TMC(Y)
    #define Y_CURRENT       800
    #define Y_CURRENT_HOME  Y_CURRENT
    #define Y_MICROSTEPS     16
    #define Y_RSENSE          0.11
    #define Y_CHAIN_POS      -1
    //#define Y_INTERPOLATE  true
    //#define Y_HOLD_MULTIPLIER 0.5
  #endif

  #if AXIS_IS_TMC(E0)
    #define E0_CURRENT      800
    #define E0_MICROSTEPS    16
    #define E0_RSENSE         0.11
    #define E0_CHAIN_POS     -1
    //#define E0_INTERPOLATE true
    //#define E0_HOLD_MULTIPLIER 0.5
  #endif

  #define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 0
  #define  Z_SLAVE_ADDRESS 0
  #define E0_SLAVE_ADDRESS 0

  #define SOFTWARE_DRIVER_ENABLE

  #define STEALTHCHOP_XY
  #define STEALTHCHOP_Z
  #define STEALTHCHOP_I
  #define STEALTHCHOP_J
  #define STEALTHCHOP_K
//  #define STEALTHCHOP_E

  #define CHOPPER_TIMING CHOPPER_DEFAULT_24V        // All axes (override below)

  #define MONITOR_DRIVER_STATUS

  #if ENABLED(MONITOR_DRIVER_STATUS)
    #define CURRENT_STEP_DOWN     50  // [mA]
    #define REPORT_CURRENT_CHANGE
    #define STOP_ON_ERROR
  #endif

  #define HYBRID_THRESHOLD

  #define X_HYBRID_THRESHOLD     100  // [mm/s]
  #define Y_HYBRID_THRESHOLD     100
  #define Z_HYBRID_THRESHOLD       15
  #define E0_HYBRID_THRESHOLD     30

  //#define SENSORLESS_HOMING // StallGuard capable drivers only
  //#define TMC_HOME_PHASE { 896, 896, 896 }

  #define TMC_DEBUG

  #define TMC_ADV() {  }

There were jumper wires that came with the drivers but I dont think they are needed with the SKR2.  Anyway, can't find where to connect them to.

I also have a SKR Turbo board on another printer that works OK...

Eddiiie, 2022-02-16:
Attached is configuraton.h and configuration_adv.h
[Configuration.txt](https://github.com/bigtreetech/SKR-2/files/8083221/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/SKR-2/files/8083222/Configuration_adv.txt)

Eddiiie, 2022-02-16:
The Ender3 is 24v.   Does that pose a problem with these drivers?

Eddiiie, 2022-02-16:
>>>m211
SENDING:M211
  M211 S1 ; ON
  Min:  X0.00 Y0.00 Z0.00   Max:  X235.00 Y235.00 Z250.00
>>>m122
SENDING:M122
		X	Y	Z	E
Address		0	0	0	0
Enabled		false	false	false	false
Set current	650	800	800	800
RMS current	642	795	795	795
MAX current	905	1121	1121	1121
Run current	20/31	25/31	25/31	25/31
Hold current	10/31	12/31	12/31	12/31
CS actual	10/31	12/31	12/31	12/31
PWM scale
vsense		1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	false
msteps		16	16	16	16
interp		true	true	true	true
tstep		max	max	max	max
PWM thresh.	98	98	131	263
[mm/s]		100	100	15	30
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
pwm scale sum	12	14	14	14
pwm scale auto	0	0	0	0
pwm offset auto	36	36	36	36
pwm grad auto	14	14	14	14
off time	0	0	0	0
blank time	24	24	24	24
hysteresis
 -end		2	2	2	2
 -start		1	1	1	1
Stallguard thrs	0	0	0	0
uStep count	40	40	40	40
DRVSTATUS	X	Y	Z	E
sg_result	0	0	0	0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:0A:00:00
		Y	0xC0:0C:00:00
		Z	0xC0:0C:00:00
		E	0x80:0C:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed
[ERROR] Can't write to printer (disconnected?) (SerialException): WriteFile failed ([Error 22] The device does not recognize the command.)

radek8, 2022-02-16:
This board was produced in 3 variants and each has its own configuration.
What processor do you have? Are you compiling FW for the right processor?

Eddiiie, 2022-02-16:
something just popped on the board.. after the m122 command above.  I didnt see what.    Something around the hot end connector on the main board I think.
dammit

radek8, 2022-02-16:
Mosfet?

Eddiiie, 2022-02-16:
[image]

I think around here...

radek8, 2022-02-16:
what stopped working

Eddiiie, 2022-02-16:
I dont know.  afraid to turn it back on at the moment.   switched to usb power and the display says TMC CONNECTION ERROR but  I can see how that can be expected since they probably dont run on USB.

Whatever popped, it probably popped open, right?  :p

```
>>>m122
SENDING:M122
		X	Y	Z	E
Address		0	0	0	0
Enabled		false	false	false	false
Set current	650	800	800	800
RMS current	1160	1436	1436	1436
MAX current	1636	2025	2025	2025
Run current	20/31	25/31	25/31	25/31
Hold current	10/31	12/31	12/31	12/31
CS actual	0/31	0/31	0/31	0/31
PWM scale
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	false	false	false	false
msteps		256	256	256	256
interp		false	false	false	false
tstep		0	0	0	0
PWM thresh.	1581	1581	2108	4216
[mm/s]		100	100	15	30
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
pwm scale sum	0	0	0	0
pwm scale auto	0	0	0	0
pwm offset auto	0	0	0	0
pwm grad auto	0	0	0	0
off time	0	0	0	0
blank time	16	16	16	16
hysteresis
 -end		-3	-3	-3	-3
 -start		1	1	1	1
Stallguard thrs	0	0	0	0
uStep count	0	0	0	0
DRVSTATUS	X	Y	Z	E
sg_result	0	0	0	0
stst		*	*	*	*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW

```

Eddiiie, 2022-02-16:
I think it was either one or all the TMC Drivers blew up.   Removing one at a time and smelling, they all seem to have a burnt smell.
Looking at the underside of the Z driver I see burn mark...
[image]

Eddiiie, 2022-02-16:
RUN AWAY FROM THIS BOARD!!!!

closed 2022-02-16 by Eddiiie

radek8, 2022-02-16:
The SKR2 revA board had a bad mosfet and if the wiring was not modified, the TMC controls burned out.
Looks like you have rev A

radek8, 2022-02-16:
https://github.com/bigtreetech/SKR-2/issues/17

This protects the TMC drivers in case you connect them upside down.
Unfortunately, they can now burn out even if they are connected properly ....

Eddiiie, 2022-02-16:
Supposedly I have a Rev B, despite the missing second sticker but having the correct MOSFET.

Returning this board. 

A real pisser.

[image]