bigtreetech/SKR-2 #123: SKR-2 TMC Error

mfaccin, 2022-02-18:
Hi, I was using the SKR-2 board for a while and a couple of days ago I started to receive the warning `Z driver backwards` in the LCD Screen.

Checking the TMC Debug gcode I have:

```
Send: M122
Recv: 		X	Y	Z	E
Recv: Address		0	0	0	0
Recv: Enabled		false	false	false	false
Recv: Set current	650	650	650	800
Recv: RMS current	642	642	1160	795
Recv: MAX current	905	905	1636	1121
Recv: Run current	20/31	20/31	20/31	25/31
Recv: Hold current	10/31	10/31	10/31	12/31
Recv: CS actual	10/31	10/31	0/31	12/31
Recv: PWM scale
Recv: vsense		1=.18	1=.18	0=.325	1=.18
Recv: stealthChop	true	true	false	false
Recv: msteps		16	16	256	16
Recv: interp		true	true	false	true
Recv: tstep		max	max	0	max
Recv: PWM thresh.	98	98	2108	274
Recv: [mm/s]		100	100	15	30
Recv: OT prewarn	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false
Recv: pwm scale sum	12	12	0	14
Recv: pwm scale auto	0	0	0	0
Recv: pwm offset auto	36	36	0	36
Recv: pwm grad auto	14	14	0	14
Recv: off time	4	4	0	4
Recv: blank time	24	24	16	24
Recv: hysteresis
Recv:  -end		2	2	-3	2
Recv:  -start		1	1	1	1
Recv: Stallguard thrs	0	0	0	0
Recv: uStep count	8	8	0	8
Recv: DRVSTATUS	X	Y	Z	E
Recv: sg_result	0	0	0	0
Recv: stst				*
Recv: olb					*
Recv: ola					*
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0A:00:00
Recv: 		Y	0xC0:0A:00:00
Recv: 		Z	0x00:00:00:00	 Bad response!
Recv: 		E	0x80:0C:00:C0
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... OK
Recv: ok
Recv:  T:26.96 /0.00 B:26.99 /0.00 @:0 B@:0
Recv:  T:27.07 /0.00 B:26.97 /0.00 @:0 B@:0
Recv:  T:26.95 /0.00 B:27.10 /0.00 @:0 B@:0
```

I already replaced all the positions of the drivers and they work fine in another spot. Is there any workaround for it? Would be possible to use the E1 as a spare channel for the Z-axis?

radek8, 2022-02-19:
Yes, just overwrite the pin file

mfaccin, 2022-02-21:
could you guide me to do it? @radek8

radek8, 2022-02-21:
I assume you won't use sensorless guidance.

radek8, 2022-02-21:
Here you have a modified pin file.
This is a file from the latest version of Marlin with bug fixes
Copy and overwrite the original file. Compile the new FW.
Connect the Z-axis controller and motor to connector E1