bigtreetech/SKR-2 #125: SKR 2 Bricked

AfroPsycho, 2022-02-22:
Hey there 
I probably killed my skr 2 while trying to connect it with raspberry via ttl uart 
The problem is I have connected raspberrys gpio pins (5v, gnd, tx, rx) to SWD pins instead of TFT pins by mistake
After this the green power led (I am not sure if it is power or not) stayed on but dimmer 
From now on I cant flash firmware via sd I cant communicate using usb b port green led is still dim but lighting plugging in a sd card lights up another green led but nothing more 

I am a student in turkey and interested in 3d stuff but because of the economics in turkey it is really hard to purchase a card like this and I don't have enough budget to purchase another one 
if this card is dead I am gonna die with it ^^
So if there is anyway you can think of or any solutions I can even modify the card I have soldering skills maybe bypassing something will make it work again?
What I can think of is when I supplied 5v and gnd to wrong pins either I have fried something or triggered some kind of protection but not sure of this 

Maybe just bad firmware?
I have complied it myself but it was working when I tried it before hooking the raspberry
Maybe boot loader got bricked

is there a way to rescue the card 
is it dead?

The picture bellow shows where I connected the dpi
[image]

AfroPsycho, 2022-02-22:
> I just checked the schematics. That doesn't sound good. You probably powered the ST with 5V. Too much, the chip dies.
> Can you check if R19 is still ok?
> Wish you the best.

There is too many resistors and I dont know which one is R19
So I have tested ones around SWD and I think I have found the R19 :’)
I have written the readings on resistors

[image]

Do we consider it as dead?

radek8, 2022-02-22:
You connected 5V to 3, 3V supply.
The processor is powered by 3.3V as written by @hcet14,
it's a lot on the processor.
Check the 3.3V supply if it works.

Try to measure 2 resistors that should have a value of 0 ohms.

[image]

hcet14, 2022-02-22:
@AfroPsycho 
Does the serial port pop up on your PC, when you connect over USB? You may try it with no power supply -> powered by USB.

AfroPsycho, 2022-02-23:
> You connected 5V to 3, 3V supply. The processor is powered by 3.3V as written by @hcet14, it's a lot on the processor. Check the 3.3V supply if it works.
> 
> Try to measure 2 resistors that should have a value of 0 ohms.
> 
> [image]

Both resistors are okay 
I have managed to read ~3.25v on 3.3v pin ( powered with usb power via pc )
So long the only bad resistor is that one around SWD pins ( one marked as 25Ohms )

AfroPsycho, 2022-02-23:
> @AfroPsycho
> Does the serial port pop up on your PC, when you connect over USB? You may try it with no power supply -> powered by USB.

I have tried to power it both from 12v and usb
Either way no connection with pc over usb

AfroPsycho, 2022-02-23:
> [image]

Hey thanks for the comment but I don’t really know what to look at on this table I didn’t understand this

radek8, 2022-02-23:
Maximum cpu power is allowed 4V

radek8, 2022-02-23:
It looks like a dead processor.

hcet14, 2022-02-23:
> I have tried to power it both from 12v and usb
> Either way no connection with pc over usb

Sorry man, the processor is death...

AfroPsycho, 2022-02-23:
Alright nothing to do then :’)
I am also waiting response from big three tech
if they say something different I will post it here maybe it will help someone someday

hcet14, 2022-02-27:
> Alright nothing to do then :’)

Well, you could replace the µC. The package has no cooling pad underneath, so a 100 pin TQFP package is easy to replace.

Your problem is getting a new µC. We have a worldwide chip crisis and these suckers are often used in automotive.

radek8, 2022-02-27:
The problem will get the bootloader

AfroPsycho, 2022-03-02:
> The problem will get the bootloader

Well this could be a problem 
For the first time I can upload marlin in to it somehow but when I solder it to the card I don’t think I can make it work with sd card for new firmwares

closed 2022-03-02 by AfroPsycho