bigtreetech/SKR-2 #126: BTT SKR 2 with TMC 2226, motor vibration when running.

pratyushganguly, 2022-02-24:
Hi,
I have got a strange and annoying issue In my printer. I have a self built core XY type printer with

- BTT SKR 2 REV B
- TMC 2226 ON X,Y AND E0
- A4988 ON Z
- REPRAP DISCOUNT FULL GRAPHIC CONTROLLER.

The motor X does not run properly after XY homing, it has a jittery operation, and also mid print, it shifts layers sometimes.

Following photos and videos should spread some light on the issue

Uploading PXL_20220223_093921536.mp4…

[image]
[image]

It was not an issue in first few prints though.

https://user-images.githubusercontent.com/97211696/155528743-142e9781-b92b-49a9-a832-3d24fd6d66d8.mp4

The measures I have taken afterwards are as follows

- checked all the wiring from motherboard to motors and end stops
- checked belts and pulleys
- replaced the stepper motor
- swapped the TMC2226 drivers from different slots.
- re-flashed the firmware, as well as flashed a fresh copy of firmware afterwards.

https://user-images.githubusercontent.com/97211696/155529877-abd66304-b402-4b6b-b2f1-a92fe3528066.mp4
In the above video, the last rotation is the movement concerned. 

kindly help

pratyushganguly, 2022-02-24:
Update
- Configured E1 stepper slot for X motor in  _pins.h_ file in marlin
- swapped the TMC2226 and stepper motor wire to E1 slot

Now, 
When homing by giving command from controller board or pronterface , it is homing fine

But when I start a print, It again vibrates after X,Y homing.

Checked slicer starting code, also added new profile in Cura, everything seems fine

Kindly help

Arakon, 2022-07-03:
No idea if this ever got figured out, but I have the a very similar issue on an SKR 1.4. Extremely loud resonating, the entire frame shakes like mad. Swapped 1:1 for some older TMC2208 I have, and the noise and buzzing is instantly gone. It was layer shifting randomly, too.

MaffooClock, 2023-02-10:
Unrelated to the SKR 2, but FWIW:

I wound up here after looking for answers on why my Voron 2.4 with TMC2226's on a Manta M8P is very loud/resonant.  I've tried many, many different configuration combinations in Klipper of run current, interpolation, microsteps, StealthChop threshold, and nothing I do does anything for the noise.  It's loudest during slow movements, but not so bad during fast movement, but never silent.

When I had everything configured exactly the way it _should_ be, that's when things were the loudest.  Dropping the run current was the only thing that would reduce the noise, and that's when I'd get layer shifting, too (due to motor skipping from not enough current).

I also have an Ender 3 that uses TMC2209 drivers, and it's 100% silent.

Again, none of this has anything to do with the SKR 2, just making a point that BTT's TMC2226 might be junk.

kloboucek, 2023-06-16:
@MaffooClock Hey Matt, have you tried changing the 2226's for 2209's as you stated? What was the result like? I'm currently building Trident with Octopus board and TMC2226's and it's unpleasantly loud no matter what I try and change in settings. It's trully hard to believe that what is meant to be a successor of 2209's is possibly so darn worse than its predecessor.