bigtreetech/SKR-2 #128: all low

Cabinfourus, 2022-02-26:
I purchased this skr2 board to replace my skr 1.4 turbo. I am now on my second board as the first one failed. The replacement board has just failed during a print. The print stopped and ALL LOW is showing on my TFT35. This board was purchased on Dec 15th and it should not fail like this. I want to know what BTT is going to do. Who do I contact to get another replacement board??

Cabinfourus, 2022-02-26:
After doing some more checking, I have found the board also burned out my tmc2209 drivers!!

berkin88, 2022-02-28:
https://www.reddit.com/r/BIGTREETECH/comments/s3c2q6/comment/hsrv9r8/?utm_source=share&utm_medium=web2x&context=3

hcet14, 2022-05-03:
> After doing some more checking, I have found the board also burned out my tmc2209 drivers!!

Have you connected MGND and PGND? [https://github.com/bigtreetech/SKR-2/issues/129](url)