bigtreetech/SKR-2 #129: Conceptional design bug

hcet14, 2022-02-27:
About me, a buddy gave me a CTC 13A (Ender 3 clone) for free. That's how I got to the 3D print stuff and I'm an absolute beginner but I know a bit about electronic circuits.

I bought the board on Ebay, because I wanted a 3D board with more flash than my Anet 3D board has and a 32bit µC. Just started reading about the board, when I won the auction.

I read about blown stepper drivers. Read, it should be solved with REV B. I didn't understand the fix, they just replaced Q1 with another MOSFET. 

Then I stumbled over this https://github.com/bigtreetech/SKR-2/issues/63#issuecomment-933762958. fitch22 explains the problem very well.

Reading more issues here, I read about some strange HW behaviour in connection with stepper drivers. I'm pretty sure it's this design bug, error, or whatever you may call it in connection with DISABLE_DRIVER_SAFE_POWER_PROTECT.
But even without using DISABLE_DRIVER_SAFE_POWER_PROTECT, what will happen if this pin toggles a little bit (whatever the reason)?

**I recommend to every owner of a SKR 2 to remove Q1 and connect drain and source pins!**

I just connected the pins with a chunk of solder. 
[image]

You may also watch this https://youtu.be/OH-h1Xz2V8g

berkin88, 2022-02-28:
https://www.reddit.com/r/BIGTREETECH/comments/s3c2q6/comment/hsrv9r8/?utm_source=share&utm_medium=web2x&context=3

hcet14, 2022-03-03:
@berkin88 
Did you use DISABLE_DRIVER_SAFE_POWER_PROTECT?

berkin88, 2022-03-11:
> @berkin88 DISABLE_DRIVER_SAFE_POWER_PROTECT kullandınız mı?

yes i did but it didn't work it still burns

hcet14, 2022-03-14:
> yes i did but it didn't work it still burns

Of course it will fry. Did you understand the explanation from fitch22 https://github.com/bigtreetech/SKR-2/issues/63#issuecomment-933762958 electrically for the problem? Don't use "DISABLE_DRIVER_SAFE_POWER_PROTECT"! To be save, connect "MGND" and "PGND"!

mscreations, 2022-03-29:
I'm curious if this is what killed my SKR2 board. When I first installed my board it worked okay and I was able to move some steppers, but the next time I restarted I got All High then after another restart All Low. Then the board died altogether. I found the 3.3v regulator died. I bypassed it with another regulator which quickly got hot along with the STM32. I bought a replacement board and all new drivers but I don't want to have a repeat failure.

hcet14, 2022-03-30:
> I don't want to have a repeat failure.

"Then I stumbled over this https://github.com/bigtreetech/SKR-2/issues/63#issuecomment-933762958. fitch22 explains the problem very well."

Do you have "deep" electrical knowledge or is there a needing to explain the problem in a more easier way?

mscreations, 2022-03-30:
No, fitch22's explanation was good. I installed a jumper on my replacement board and so far it's been working okay. I asked on Discord if when we compile firmware for the SKR 2 board we should get a warning regardless of whether it is a rev a or b board. I haven't gotten a response yet but I think it would be prudent to do so. It would have saved my first board and drivers if I had seen that during compilation.

hcet14, 2022-03-30:
> Discord 

I don't know what you mean by "Discord".

> I installed a jumper on my replacement board

Where?

> get a warning regardless of whether it is a rev a or b board

Doesn't matter. It is a "Conceptional design bug".

hcet14, 2022-04-01:
> I'm on the Marlin firmware Discord server where devs can talk without necessarily doing so directly on GitHub.

Thanks, I didn't know that.

> From the board power supply negative to the ground pins on the CLS headers (pin3 on each one). The video I watched said to jump at least two which is what I did. That ties MGND to PGND.

Did you do that before you powered up for the first time with all the drivers and other stuff connected?

mscreations, 2022-04-02:
> > From the board power supply negative to the ground pins on the CLS headers (pin3 on each one). The video I watched said to jump at least two which is what I did. That ties MGND to PGND.
> 
> Did you do that before you powered up for the first time with all the drivers and other stuff connected?

Yep. I got it all wired up with the jumper in place prior to flashing the firmware and running it for the first time.