bigtreetech/SKR-2 #131: Problems with file "pins_BTT_SKR_V2_0_common.h" for EXP1 and EXP2

hcet14, 2022-03-14:
I'm trying to hook up a "A10S A13 Display" to SKR 2, EXP1.
[image]
[image]
No success so far. Trying to investigate the problem leaded me to the differences of the "pins_BTT_SKR_V2_0_common.h" file.

I think the pins should correlate to the schematics. I checked the schematics with my multimeter, schematics are fine regarding EXP1 and EXP2.

https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin/src/pins/stm32f4/pins_BTT_SKR_V2_0_common.h
[image]
#define EXP1_03_PIN                         PE13
#define EXP1_04_PIN                         PE12
#define EXP1_05_PIN                         PE11
#define EXP1_06_PIN                         PE10
#define EXP1_07_PIN                         PE9
#define EXP1_08_PIN                         PB1
#define EXP1_09_PIN                         PB0
#define EXP1_10_PIN                         PC5

#define EXP2_03_PIN                         -1
#define EXP2_04_PIN                         PC4
#define EXP2_05_PIN                         PA7
#define EXP2_06_PIN                         PB2
#define EXP2_07_PIN                         PA4
#define EXP2_08_PIN                         PE7
#define EXP2_09_PIN                         PA5
#define EXP2_10_PIN                         PA6

https://github.com/bigtreetech/SKR-2/blob/master/Firmware/Marlin-bugfix-2.0.9.2.x/Marlin/src/pins/stm32f4/pins_BTT_SKR_V2_0_common.h
[image]
#define EXP1_03_PIN                         PE13
#define EXP1_04_PIN                         PE12
#define EXP1_05_PIN                         PE11
#define EXP1_06_PIN                         PE10
#define EXP1_07_PIN                         PE9
#define EXP1_08_PIN                         PB1
#define EXP1_09_PIN                         PB0
#define EXP1_10_PIN                         PC5

#define EXP2_03_PIN                         -1
#define EXP2_04_PIN                         PC4
#define EXP2_05_PIN                         PA7
#define EXP2_06_PIN                         PB2
#define EXP2_07_PIN                         PA4
#define EXP2_08_PIN                         PE7
#define EXP2_09_PIN                         PA5
#define EXP2_10_PIN                         PA6

hcet14
[image]
#define EXP1_01_PIN                         PC5
#define EXP1_02_PIN                         PB0
#define EXP1_03_PIN                         PB1
#define EXP1_04_PIN                         PE9
#define EXP1_05_PIN                         PE10
#define EXP1_06_PIN                         PE11
#define EXP1_07_PIN                         PE12
#define EXP1_08_PIN                         PE13

#define EXP2_01_PIN                         PA6
#define EXP2_02_PIN                         PA5
#define EXP2_03_PIN                         PE7
#define EXP2_04_PIN                         PA4
#define EXP2_05_PIN                         PB2
#define EXP2_06_PIN                         PA7
#define EXP2_07_PIN                         PC4
#define EXP2_08_PIN                         -1

I'm pretty sure, mine is the right one. 

Any help is appreciated!

radek8, 2022-03-20:
I assume your display is not connected as EXP1 but EXP3.
You must use a Y-shaped cable to connect the display
which has 2 connectors on one side and 1 connector on the other.
You can buy the cable or make it yourself.

radek8, 2022-03-20:
https://www.aliexpress.com/item/1005001325478608.html?af=4043355_1&cv=39321611&cn=45r91yepl3fnzkanqgjk4bm2f9uo97tv&dp=v5_45r91yepl3fnzkanqgjk4bm2f9uo97tv&product_id=1005001325478608&afref=https%3A%2F%2Falipromo.com%2Fredirect%2Fcpa%2Fo%2Fqg8snjt3wvb72hvmhs4d92i7a8685ioh%2F&aff_fcid=d67bf1c7571c4579bb868572b67279c7-1647794545480-09201-_pvnUFuy&tt=API&aff_fsk=_pvnUFuy&aff_platform=api-new-link-generate&sk=_pvnUFuy&aff_trace_key=d67bf1c7571c4579bb868572b67279c7-1647794545480-09201-_pvnUFuy&terminal_id=290428c8106c48438ea1ffecefca338b&afSmartRedirect=y

radek8, 2022-03-20:
https://www.aliexpress.com/item/1005001451518988.html?spm=a2g0o.detail.1000014.1.3a007b008OjaKj&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40000.267768.0&scm_id=1007.40000.267768.0&scm-url=1007.40000.267768.0&pvid=fcb53994-b2ea-4d03-b2c0-e1bf8236726d&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40000.267768.0,pvid:fcb53994-b2ea-4d03-b2c0-e1bf8236726d,tpp_buckets:668%232846%238110%23382&pdp_ext_f=%257B%2522sku_id%2522%253A%252212000016164623409%2522%252C%2522sceneId%2522%253A%252230050%2522%257D&pdp_pi=-1%253B50.72%253B-1%253B-1%2540salePrice%253BCZK%253Brecommend-recommend

radek8, 2022-03-20:
You can also try:
comment
// # define A10S_A13_Display

uncomment
#define CR10_STOCKDISPLAY

Use an unmodified pin file.

hcet14, 2022-04-16:
Thanks radek8.

I didn't have much time, but now it's Easter vacation...
[image]
Choose "#define ANET_FULL_GRAPHICS_LCD_ALT_WIRING" in Configuration.h. Everything else is explained in attached pins_BTT_SKR_V2_0_common.h.
[pins_BTT_SKR_V2_0_common.zip](https://github.com/bigtreetech/SKR-2/files/8500484/pins_BTT_SKR_V2_0_common.zip)

hcet14, 2022-04-19:
No comment.

closed 2022-04-19 by hcet14