bigtreetech/SKR-2 #132: SKR2 USB connection to PC fail

iankds05, 2022-03-27:
So this is todays issue. Purchased a SKR 2 B on ebay. Came still sealed so assumed was new from BTT originally. Installed in Anycubic 4Max Pro v2 and completely rewired the harness, plus upgraded to 24V operation from existing 12v. Also have a BTT TFT35 V2 screen. Everything has been flashed with relevant current firmware, but the one thing that has eluded to work is USB connection to my PC using either Windows 10 or Windows 7. There is no indication in device manager of there being a USB device connection or a device that requires driver update. No drivers should be needed because I can connect to both my Prusa and Anet AM printers with no issues. I have trailed through all the comments and issue folks have had with connection to Octoprint thinking this may shed some light but nothing so far has materialized. My serial ports are configured as SP1 1, SP2 -1  SP3  2. I have tried other combinations also to no avail. This is the only outstanding issue I would like to solve as I would like to communicate via Prontoprint to this printer. Any glimmer of light would be helpful from anyone who may have solved a similar issue on the SKR 2 board. The other point is if the USB port is connected to the PC it reports over voltage and the Cable gets hot, so something is not correct.

hcet14, 2022-03-30:
[image]
Where is your jumper?

iankds05, 2022-03-30:
The jumper is in the VDD position because the board is fully installed and works apart from the fact it cannot communicate via USB. It will even power via USB when the jumper is changed, I suspect that the ESD diode bridge on the USB line is breaking down when plugged into the pC

hcet14, 2022-04-01:
> I suspect that the ESD diode bridge on the USB line is breaking down when plugged into the pC

I don't think so. Did you connect MGND and PGND?

hcet14, 2022-04-19:
I just ran into your problem. I'm configuring Marlin under Ubuntu. When I installed my first TMC2209, I was in Windows (dual boot PC) and tried to connect to the SKR 2 for the first time. No success, played with the com port settings, no success. Little afraid (maybe I got a bad TMC), I booted back to Ubuntu. No problems. So back to Windows.

What comes now is typical Redmond...

First I installed the ST32 driver. No success, So back to the Windows driver.

[image]
[image]

... but successful...

[image]

As you can see three different baud rates. Long live Windows...

Configuration.h>
#define SERIAL_PORT -1 
#define BAUDRATE 250000

My system:
Windows 10 Pro 21H1
Ubuntu 20.04.4 LTS
Marlin-bugfix-2.0.x from 220418

Please give a feedback if it worked.

@MarkEhr
I have the same problem, compiling and uploading the firmware.bin file to SD card in Marlin doesn't work. Since this issue is already closed [https://github.com/bigtreetech/SKR-2/issues/115#issue-1118254829](url) I write here.

I can see the SD card under Ubuntu and just copy the file there. So no SD card pulling anymore.

[image]

...but Richmond...

[image]

I'm going to open an issue, but am not sure where. PlatformIO or OpenOCD? Probably not Marlin. What do you think?

iankds05, 2022-04-20:
Well I dug a little deeper into this issue, having replaced my board with an MKS Nano V3. I figured there had to be a reason the USB cable got hot. Turns out the protection diode bridge across the D=/D- lines was destroyed and basically turned into a low ohmic resistor. I ordered up a new one form Aliexpress and with a bit of persuasion got the old one out without destroying the tracks. It solved the shorting issue but I guess one of the inner layers of the PCB was damaged so I had to solder a jumper wire from the MCU pins to the USB socket. Its all good now, so we will see how long this MacGyver mod works before something else dies on this board. I'm still not impressed with this board and will just keep it as a backup, because I don't trust the design long term. I wont be buying any BTT products any time soon, so they lost me on this one.  If you are going to raise this issue maybe it should be on the BTT web sites or github so others looking for this type of issue can find it. I doubt it will be addressed any time soon by BTT as it strikes me this company is just churning out stuff to fill the coffers and not that interested in after sales service. I won't be supporting that sort of company

hcet14, 2022-04-20:
>  I guess one of the inner layers of the PCB was damaged so I had to solder a jumper wire from the MCU pins to the USB socket. 

Thanks to the electronic god, you have been lucky in the end.

> I'm still not impressed with this board and will just keep it as a backup, because I don't trust the design long term. I wont be buying any BTT products any time soon, so they lost me on this one.

I totally agree and BTT is not really willing to help. I wrote the following to BTT support:  I guess it is a 4 layer layout, I'm interested in the 2 inner layers. The answer was:  Some of the hardware and software materials we can provide can be found on github. The circuit design of the motherboard is not open source.
I wrote back: Thank you for nothing. That makes it hard for me to improve your buggy product for free.
I just wanted to know, where the MGND and PGND planes are. So nothing big.

> If you are going to raise this issue maybe it should be on the BTT web sites or github so others looking for this type of issue can find it. 

The issue would be about the uploading problem in AutoBuildMarlin. That's not a BTT fault. I guess PlatformIO or OpenOCD. What do you think? Did you ever tried the automatic upload function in PlatformIO?