bigtreetech/SKR-2 #133: SKR 2 Exp2 Activating NC Pin?

AfroPsycho, 2022-03-28:
Hey I am upgrading from MKS Gen 1.4 to SKR 2 
in mks board encoder button was located on D41 in Marlin
With the same wiring there is no corresponding pin on SKR 2 
I don't want to change the wiring

is it possible to activate that nc pin? 

[image]

[image]

hcet14, 2022-03-30:
> it possible to activate that nc pin?

No, NC means "not connected", sorry.

AfroPsycho, 2022-03-30:
> > it possible to activate that nc pin?
> 
> No, NC means "not connected", sorry.

Oh, I thought it was something like normally closed 
Anyways thanks for answer

closed 2022-03-30 by AfroPsycho