bigtreetech/SKR-2 #136: External Reset bricks the SKR2

marcossfernandez, 2022-04-13:
Putting an external reset button, connected to EXP2 on GND and RESET causes the board to no longer respond. I think it bricks them.

It is a Normally Open momentary push button, which closes the two pins when pressed, just like the board's reset button does.

Why does it behave differently? How to safely have a external reset switch?

It'a a SKR 2 Rev B 429 chip.

hcet14, 2022-04-16:
Strange, I just did it with a pair of pincers and my F429 resets perfectly.

Maybe you chose a terrible bouncing push button?

marcossfernandez, 2022-04-25:
I can confirm it's not related with the reset. Another board without external reset, suffer the same problem. 

The USB connection to RPI (or a PC) was missing after some hours of use. Just powered off and when back on some minutes after, the board can't stablish any comunication to the RPI (or PC) anymore. The /dev/serial/by-id/usb-STMicroelectronics_STM32F429VGT6 is missing afterwards.
Now we have 3 SKR 2 boards that we suposse it can works alone but they can't use the USB communication port. Our setup use Repetier server on a Raspberry PI to interface with the SKR thru USB, so there are unuable. Already asked BTT for replacement.

Found somewhere this could be caused by a defective diode in D+/D- on USB data circuit.

marcossfernandez, 2022-05-04:
We are tired of test SKR2 boards. All of them suffer the same problem, after some time working fine and tested ok (movements, extrusions, etc) they bricked, after a reset, after a communication timeout or hang, always ends bricked.
After that the board acepts firmware update from the SD but no USB comunication at all.

The most frustrating thing of all is that you don't know what causes the failure, what components are damaged and why it happens.
We know enough about electronics to see that this board is not well designed and the components or the PCB itself are not of high quality. 

It is not normal in a week 4 identical printers, that they work fine one or two days with this boards and after switching them on, they suddenly stop working. It's not external related, all pheriperals, connectors, cables and power supplies are absolutely fine.

We have had 8 units so far. It is also a waste of money.

Anyone could help us to troubleshoot and seek for the failed components. Replace it isn't enough! We need to know why these boards fail!
[image]

marcossfernandez, 2022-05-04:
> 
The external reset is fine, with other controllers works well. And if its the case, there's attached a 0.1uF capacitor for mitigate that problem.

The problem also happen when the internal reset it's triggered. In fact GND and RESET pins are in paralell of the SMD reset onboard.

hcet14, 2022-05-07:
> We are tired of test SKR2 boards.

I understand that, but keep fighting. You spend a lot of money.
First of all, did you connect MGND and PGND? ["https://github.com/bigtreetech/SKR-2/issues/129"](https://github.com/bigtreetech/SKR-2/issues/129)

marcossfernandez, 2022-05-09:
More than the money spent, it is the bad reputation due to a faulty SKR2 boards and the inconvenience caused to customers as a result, that annoys us the most.

And yes, the boards are connected at the back of the PGND to the MGND. They have continuity bypassing the HY mosfet.
We have also enabled Marlin firmware to not use the anti-SNAFU function. 

However, one of the modified boards has still failed. I can't trust BTT anymore.