bigtreetech/SKR-2 #137: SKR 2 Failed to connect to PC and Firmware Doesn't Flash!

BetulSBoyali, 2022-04-15:
Hello, 
The HE0 thermistor in the heatblock was got out of its hole. When I heated up, some smoke came out of hotend, and gave me a heating error HE0..
But then I lost connection with my SKR 2. My TFT35 seems like it has no connection with firmware, it doesn't show any of the thermistor's value, the motors do not move! 
I decided flashing new firmware, thats how I burned my SD card. It doesnt work anymore.. And when i was pulling the sd card, I got tickled by electric from sd card.
I bought SMD fuses after I came across some experiences from people, I thought they might be dead, I replaced them but nothing works..
I tried to flash firmware with new sd card, it did not die this time though.

Please I need help on this! @bigtreetech

hcet14, 2022-04-16:
Is there any LED on, when powered? You could try with the simplest HW configuration. Disconnect everything and I mean everything from your SKR 2 (like here without display and SD card [https://user-images.githubusercontent.com/38034498/163688030-0784f243-740f-40bd-a865-50102800f302.jpg](url) ) . Set the jumper to USB power (same on the picture) and connect to USB to your computer. Does the serial USB device appears on your computer?

Good luck, hcet14

BetulSBoyali, 2022-04-16:
> Is there any LED on, when powered? You could try with the simplest HW configuration. Disconnect everything and I mean everything from your SKR 2 (like here without display and SD card [...]

Thank you for the reply @hcet14 
I will try disconnecting everything but I can't view the pic you attached. There is yellow led near SD card socket that is on, but it doesn't blink. Also the orange leds are on near RGB sockets. But there is no response of flashing the firmware at all. 
I will try disconnecting and write here. Thank you

BetulSBoyali, 2022-04-16:
I disconnected everything.. no response of firmware flashing except leds are on, they dont blink at all. It used to load firmware.bin and change to FIRMWARE.CUR but it doesn't do that anymore either.

hcet14, 2022-04-19:
> I disconnected everything.. no response of firmware flashing except leds are on, they dont blink at all. 

That's good. Two red Leds and a green one, when your SD card is in the slot. The left LED comes from the board (D20), so it is probably alive ;)

> It used to load firmware.bin and change to FIRMWARE.CUR but it doesn't do that anymore either.

That comes later, first you could check if the USB connection still exists. Open the Windows Device Manager (I'm guessing you are using Windows) and check if it looks like this. When you connect the SKR 2 to USB a "Serial USB Device" should show up

[image]

BetulSBoyali, 2022-04-21:
> That's good. Two red LEDs and a green one, when your SD card is in the slot. The left red LED comes from the board (D20), so it is probably alive ;)
> >Yes, exactly as what you said. All three are on, but 

> That comes later, first you could check if the USB connection still exists. Open the Windows Device Manager (I'm guessing you are using Windows) and check if it looks like this. When you connect the SKR 2 to USB a "Serial USB Device" should show up
> Actually I did this earlier, I checked it there is no ''serial USB device'' connection shown in the list. Before, there used to be connection tone to PC, that doesn't come anymore. No connection at all...

> 
> Oh, one thing I forgot to ask. Is your SD card still working in your PC? Are you able to format it? Not a "quick format", a full format?
Yes, I just checked, it did format fully. And it is working in my pc. The problem is SD card doesnt flash the firmware and no connection between SKR2 and PC.. I also checked with another USB ports and cable to make sure if any of the port or the cable is dead, but still no sign of connection.

hcet14, 2022-04-21:
> And when i was pulling the sd card, I got tickled by electric from sd card.

Was the power plug in the wall socket?

When you have the device manager open and you connect SKR and PC. Does it enumerate? I.e does the device manager flashes to white for a second?

MarvinLind, 2022-04-23:
I have the same issue. TH0 started acting up, got a heat warning, changed the probe and now the LCD is blank, no serial via USB or RX/TX(PA9/10). And firmware won't flash from SD-card...

hcet14, 2022-04-25:
What is M122 saying?

MarvinLind, 2022-04-25:
> What is M122 saying?

Sorry, without serial this question is hard to answer.

MarvinLind, 2022-04-25:
> I have the same issue. TH0 started acting up, got a heat warning, changed the probe and now the LCD is blank, no serial via USB or RX/TX(PA9/10). And firmware won't flash from SD-card...
> 
> Edit: Now that I think of it. It did boot a couple of times but said something like "TMC Communication Error", I removed all TMC-drivers to inspect them but there was nothing strange. After that the above issues started.

This might have something to do with me trying to flash my EEPROM module earlier that day. Is it possible that I erased the bootimage? However this was done ~12 hours earlier and it still booted after that (but in the end with the TMC issues). And even later no boot at all (but leds are active).

I can't find a bootimage for Rev B (F429) so can't really test to re-flash it.

hcet14, 2022-04-25:
> Sorry, without serial this question is hard to answer.

Stupid me, sorry ;)

> I can't find a bootimage for Rev B (F429) so can't really test to re-flash it.

I have no programming knowledge with the STM32 µCs, but I guess you need a ST-Link. You can buy cheap clones on Ebay or Aliexpress.

BetulSBoyali, 2022-04-25:
> > And when i was pulling the sd card, I got tickled by electric from sd card.
> 
> Was the power plug in the wall socket?
> 
It was in triple power socket and that was connected to wall socket.

> When you have the device manager open and you connect SKR and PC. Does it enumerate? I.e does the device manager flashes to white for a second?
>
No it doesn't. No sign of connection..

MarvinLind, 2022-04-26:
> > Sorry, without serial this question is hard to answer.
> 
> Stupid me, sorry ;)

No problems mate :)

> > I can't find a bootimage for Rev B (F429) so can't really test to re-flash it.
> 
> I have no programming knowledge with the STM32 µCs, but I guess you need a ST-Link. You can buy cheap clones on Ebay or Aliexpress.

I have the equipment but not the boot image for Rev B. Can find it for rev A but I'm guessing it's not the same (haven't tried)

BetulSBoyali, 2022-04-26:
@MarvinLind any solution so far? 

Because I cant still use my SKR 2 anymore because no connection.

hcet14, 2022-04-27:
> I have the equipment but not the boot image for Rev B. Can find it for rev A but I'm guessing it's not the same (haven't tried)

Are you able to communicate to your STM32 by ST-Link? Again, there is no difference between Rev A and Rev B. They just replaced Q1. All the guys here complaining about strange behavior of their drivers didn't do this, connect MGND and PGND.

[image]
See also [https://github.com/bigtreetech/SKR-2/issues/129#issue-1153264600](url)
If your communication is established, you could install ( boot image for Rev A), I guess you talk about the bootloader?

hcet14, 2022-04-27:
> It was in triple power socket and that was connected to wall socket.

That's good news. I think the "tickled by electric" came from you (ESD) not from your sd card. If it was ESD, the plugged in power cord is helping to prevent damage of your STK 2.
Maybe MarvinLind makes some progress, getting that µC working again.

MarvinLind, 2022-05-07:
> > I have the equipment but not the boot image for Rev B. Can find it for rev A but I'm guessing it's not the same (haven't tried)
> 
> Are you able to communicate to your STM32 by ST-Link? Again, there is no difference between Rev A and Rev B. They just replaced Q1. All the guys here complaining about strange behavior of their drivers didn't do this, connect MGND and PGND.
> 
> [image] See also [https://github.com/bigtreetech/SKR-2/issues/129#issue-1153264600](url) If your communication is established, you could install ( boot image for Rev A), I guess you talk about the bootloader? But I have no experience with that. Good luck! If I'm lucky, I have some old ST32 development boards somewhere. They should have an ST-Link onboard. Maybe I try to establish a connection over the weekend.

I've given it some tries now but are not able to read from memory. It gets the Device name and ID but I can't read or write to it. Trying to set the option bytes to R/W fails. Tested this under linux and windows with different drivers but I don't think that is the issue as I get name and ID out.

hcet14, 2022-05-07:
> I've given it some tries now but are not able to read from memory. 

ouupss

> It gets the Device name and ID but I can't read or write to it. Trying to set the option bytes to R/W fails. 

Which software tools are you using?

> Tested this under linux and windows with different drivers but I don't think that is the issue as I get name and ID out.

Ok, we can eliminate that!

Have you had a look at this? ["https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader"](https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader)

I'm working at it. I found an ST-Link in my garbage, but I'm also struggling with the SW from ST. I didn't spend much time for that, since I have little right now, sorry.

Another thing is, I have a working SKR 2! I'm looking for a messed up SKR 2 to buy for a good price. I want a backup board when I start messing with the FW! Is there anybody here in this forum, who is willing to sell a messed up board for a good price? My location is Germany. Soldering is no problem, I have sophisticated soldering and watching tools available at work. I will not replace a µC on the board (too time consuming)! So the µC should be alive! It would be nice, if I could buy a SKR 2 with an F429.

MarvinLind, 2022-05-08:
> > I've given it some tries now but are not able to read from memory.
> 
> ouupss
> 

Haha indeed...

> Which software tools are you using?
>
>
> Ok, we can eliminate that!
>
> Have you had a look at this? [...]

Yes GadgetAngels methods and guides are the ones I've followed. Unfortunately he doesn't cover the SKR 2 in his documentation so I don't know if there is something special I need to do beforehand. He mentions that on SKR E3 (or some board similar) you have to boot a special version of Marlin for the SWD connection to work.

> I'm working at it. I found an ST-Link in my garbage, but I'm also struggling with the SW from ST. I didn't spend much time for that, since I have little right now, sorry.
> 
> Another thing is, I have a working SKR 2! I'm looking for a messed up SKR 2 to buy for a good price. I want a backup board when I start messing with the FW! Is there anybody here in this forum, who is willing to sell a messed up board for a good price? My location is Germany. Soldering is no problem, I have sophisticated soldering and watching tools available at work. I will not replace a µC on the board (too time consuming)! So the µC should be alive! It would be nice, if I could buy an SKR 2 with an F429.

Well I have mine that doesn't work (Rev B) and if I don't manage to fix it I'll think about it/you.

Thank's for your reply!

hcet14, 2022-05-08:
> Well I have mine that doesn't work (Rev B) and if I don't manage to fix it I'll think about it/you.

Alright, don't get me wrong. I still  keep my fingers crossed for your success!

MarvinLind, 2022-05-12:
> > Well I have mine that doesn't work (Rev B) and if I don't manage to fix it I'll think about it/you.
> 
> Alright, don't get me wrong. I still keep my fingers crossed for your success!

I got a refund and sent it back to Amazon. New board arrived today and without flashing it with Marvin no SWD was possible. After flashing it reading the bootloader and marlin firmware was possible. I'm not going to try any writes to it via SWD but are gonna make a hex-dump of the full sram,

hcet14, 2022-05-14:
> I got a refund and sent it back to Amazon.

 Very good! :thumbsup: 

> I don't manage to fix it I'll think about it/you.

too bad for me! :smile:

This might be interesting ["https://blog.cyril.by/en/software/debugging-marlin-with-st-link"](https://blog.cyril.by/en/software/debugging-marlin-with-st-link)

We get the SKR2 working!