bigtreetech/SKR-2 #138: Wire Bltouch v3 using zmin

davbcr, 2022-04-19:
I'm having hard times trying to properly wire the bl touch!
I cannot use servo pins because I've another servo there.
[image]
[image]
BLTouch comes with a 3 + 2 wires connections that I hoped corresponds to skr2 connections and i connecter 
Yellow-PC0
White-PE4
using the ground close to PE4 for black ground wire and 5v and ground from Z stop port for red and brown,but it doesn't work.
BLTouch keeps permanently red after the initial check as the printer is switched on.
I'm starting to think to connect white-PC0 and yellow-PE4.
Do i get it correctly lastly?

hcet14, 2022-04-19:
Please post a picture of your connections. I guess that makes it clearer.

davbcr, 2022-04-20:
> Please post a picture of your connections. I guess that makes it clearer.
> 
> But you have a 3D Touch, don't you?

I'm not so confident it'll be clearer but here they are and Yep 3D touch.
[image]
[image]
[image]

hcet14, 2022-04-20:
> I'm not so confident it'll be clearer but here they are and Yep 3D touch.

Well, still confusing. Could you paint it into BIGTREETECH SKR 2-SCH.pdf or BIGTREETECH SKR 2-Pin.pdf?
I'm planning to use a 3D touch in the future, so I'm interested how you get that thing working.

davbcr, 2022-04-20:
Please forgive me for confusion.
These are the connections i made, but the sensor doesn't work and keeps staying red after the initial check when you switch the printer on:
[image]

Do i have to connect this way instead?
[image]

hcet14, 2022-04-20:
Ok, now I got it.

> Do i have to connect this way instead? 

Give me some time to think about it. I wouldn't choose the z-stop pins since they are so involved in Marlin. What other pins are available? Choose some Marlin doesn't use. That's the good thing with this board, many IOs. Have a look in the datasheet of the STM32. The pins you choose should tolerate 5V. Some don't and you may brake the µC.
Also have a look in the pins_BTT_SKR_V2_0_common.h, since you have to mess with that one. The z-stop-PROBE stuff is at the very end.
I'll get back to you, right now I got no time. 
Write to you later.

radek8, 2022-04-20:
If you want to use other pins for the servo, you must select a pin that supports PWM control.
The Z-stop connector is not suitable, a pullup resistor is connected to it.
Use an RGB connector (which supports PWM) for brown, red and yellow wires. Probe connector for white and black wire.
You must also define the servo pin in the pin file.

radek8, 2022-04-20:
The servo index for BlTouch is not defined anywhere in the configuration files. So I assume that servo0 is assumed by default.
So you should modify the configuration and define BlTouch as SERVO0 and the second servo as SERVO1.

davbcr, 2022-04-21:
> If you want to use other pins for the servo, you must select a pin that supports PWM control. The Z-stop connector is not suitable, a pullup resistor is connected to it. Use an RGB connector (which supports PWM) for brown, red and yellow wires. Probe connector for white and black wire. You must also define the servo pin in the pin file. #define SERVO1_PIN PE6
> 
> I have a SKR 1.4 board and I also use 2 servos. I have 3DTouch connected to the servo + probe connector. The second servo is in the RGB connector. I defined the servo in the pin file and both servos work flawlessly.

Due because i've also the Led strip connected there what you think about using the FAN2 PB5 pin for yellow wire instead?
[image]
It should be a PWM pin,right?
so i can #define SERVO1_PIN PB5 in the pin file, and #define Z_PROBE_SERVO_NR 1 in the config.h, right?

radek8, 2022-04-21:
I don't know if the value #define Z_PROBE_SERVO_NR
will also affect BLTOUCH. You would have to try that.
But it is not a problem for BLTouch to be on the servo with index 0 and the second servo on index1. This is just about configuration.

Yes the PB5 pin could be used, but not from the FAN2 connector.
the output from the Mosfet transistor is connected to the FAN2 connector, which switches GND to the fan.
You would have to pull pin PB5 from the Gate transistor.

What takhleled lights cancel?
Do you want to build a printer with leveling or a disco ball with colorful music? :-)

davbcr, 2022-04-21:
Ok gottcha, both of you. Thanks a lot.
I will proceed as Redek suggested:
- BLTOUCH connected to SERVO and PROBE pins 
- second servo as SERVO1 to RGB ports
I'll try to set it up this way and report back if it's working.

hcet14, 2022-04-21:
@radek8 
> you must select a pin that supports PWM control

I don't think so. Every pin is capabable to output the software generated PWM signal.  The dedicated PWM pins are just used, if the hardware in the CPU is used to generate the PWM signal. 

@davbcr 
> BLTOUCH connected to SERVO and PROBE pins
> second servo as SERVO1 to RGB ports

Makes sense. Much easier to configure Marlin. What is the other servo used for? Good luck!

radek8, 2022-04-21:
> @radek8
> 
> > you must select a pin that supports PWM control
> 
> I don't think so. Every pin is capabable to output the software generated PWM signal. The dedicated PWM pins are just used, if the hardware in the CPU is used to generate the PWM signal.
> 
 @hcet14 
I don't want to argue, I don't have such a good knowledge of it, but even on Arduino, only some pins support PWM.
They are marked ~

[image]
https://create.arduino.cc/projecthub/muhammad-aqib/arduino-pwm-tutorial-ae9d71

radek8, 2022-04-21:
List of pins supporting PWM on processor STM32F4...

https://stm32f4-discovery.net/2014/09/library-33-pwm-stm32f4xx/

davbcr, 2022-04-21:
> @davbcr
> 
> > BLTOUCH connected to SERVO and PROBE pins
> > second servo as SERVO1 to RGB ports
> 
> Makes sense. Much easier to configure Marlin. What is the other servo used for? Good luck!

It's used for the servo that control the idler movement of the MultiMaterialUnit designed by Pierre Masselot, he did really a great job creating a MMU kit you can very easily print and assemble for much less than 100€.
I already did and assembled it but i need the SKR2 full 100% functioning before getting to fine tune it and finally print with multi material or different colors. :D

hcet14, 2022-04-22:
> I don't want to argue

No, you are absolutely right! 

This is from the F429 datasheet:

> 3.22.1
> STM32F427xx STM32F429xx
> Advanced-control timers (TIM1, TIM8)
> The advanced-control timers (TIM1, TIM8) can be seen as three-phase PWM generators
> multiplexed on 6 channels. They have complementary PWM outputs with programmable
> inserted dead times. They can also be considered as complete general-purpose timers.
> Their 4 independent channels can be used for:
> • Input capture
> • Output compare
> • PWM generation (edge- or center-aligned modes)
> • One-pulse mode output
> If configured as standard 16-bit timers, they have the same features as the general-purpose
> TIMx timers. If configured as 16-bit PWM generators, they have full modulation capability (0-
> 100%).
> The advanced-control timer can work together with the TIMx timers via the Timer Link
> feature for synchronization or event chaining.
> TIM1 and TIM8 support independent DMA request generation.
> 3.22.2
> General-purpose timers (TIMx)
> There are ten synchronizable general-purpose timers embedded in the STM32F42x devices
> (see Table 6 for differences).
> •
> TIM2, TIM3, TIM4, TIM5
> The STM32F42x include 4 full-featured general-purpose timers: TIM2, TIM5, TIM3,
> and TIM4.The TIM2 and TIM5 timers are based on a 32-bit auto-reload
> up/downcounter and a 16-bit prescaler. The TIM3 and TIM4 timers are based on a 16-
> bit auto-reload up/downcounter and a 16-bit prescaler. They all feature 4 independent
> channels for input capture/output compare, PWM or one-pulse mode output. This gives
> up to 16 input capture/output compare/PWMs on the largest packages.
> The TIM2, TIM3, TIM4, TIM5 general-purpose timers can work together, or with the
> other general-purpose timers and the advanced-control timers TIM1 and TIM8 via the
> Timer Link feature for synchronization or event chaining.
> Any of these general-purpose timers can be used to generate PWM outputs.
> TIM2, TIM3, TIM4, TIM5 all have independent DMA request generation. They are
> capable of handling quadrature (incremental) encoder signals and the digital outputs
> from 1 to 4 hall-effect sensors.
> •
> TIM9, TIM10, TIM11, TIM12, TIM13, and TIM14
> These timers are based on a 16-bit auto-reload upcounter and a 16-bit prescaler.
> TIM10, TIM11, TIM13, and TIM14 feature one independent channel, whereas TIM9
> and TIM12 have two independent channels for input capture/output compare, PWM or
> one-pulse mode output. They can be synchronized with the TIM2, TIM3, TIM4, TIM5
> full-featured general-purpose timers. They can also be used as simple time bases.

The according file where the PWM in Marlin is done should be fast_pwm.cpp.

But that is something I would never touch. The files I touch are platformio.ini, Configuration.h, Configuration_adv.h, and pins_BTT_SKR_V2_0_common.h.

So I would say we just route an output pin to another pin in pins_BTT_SKR_V2_0_common.h.

hcet14, 2022-04-22:
> MultiMaterialUnit designed by Pierre Masselot

Ok, interesting. Good luck!

closed 2022-05-02 by davbcr