bigtreetech/SKR-2 #139: Nozzle to probe offset being ignored

bohomurcek, 2022-04-21:
I recently updated my X5sa to SKR2 mainboard.
Everything works fine except nozzle to probe offset being ignored. 
My probe is to the left and front and I am 100% sure my settings are right.
#define NOZZLE_TO_PROBE_OFFSET { -49, -8, 0 }

Any input is highly apreciated.

radek8, 2022-04-23:
Send configuration files and video with faulty cation

bohomurcek, 2022-04-24:
[image]
[image]
Hi radek8, 

thank you for willingness to check my config. Kindly find files attached.
I did not make a video since for the temporary solution i used "#define PROBING_MARGIN 50", so problem with probing outside of bed would not be visible on video except for the smaler bed probing.

Another intersting thing about the problem is that I can see the probe offset in Marlin mode correctly but in
touch mode is 0 and if I set it, it resets back to zero. On the other hand Z-offset works flawlessly.
Same, if I use [M851] it returns 0 even after setting otherwise with it. As a probe I use optic switch conected as a Zmin probe.

Kind Regards

radek8, 2022-04-24:
Do you use current FW for TFT?

bohomurcek, 2022-04-24:
Yes I did.

radek8, 2022-04-24:
What is the creation date of FW TFT?

radek8, 2022-04-24:
You have defined
#define FIX_MOUNTED_PROBE
It says that it is a nozzle-based probe.
If the probe is on the nozzle, then the probe offset is 0.

`/**
 * A Fix-Mounted Probe either doesn't deploy or needs manual deployment.
 *   (e.g., an inductive probe or a nozzle-based probe-switch.)
 */
#define FIX_MOUNTED_PROBE`

bohomurcek, 2022-04-24:
It is 26. Jan

radek8, 2022-04-24:
If you see offset correctly in Marlin and incorrectly on TFT, you have an old FW on TFT.
Upgrade to the new FW TFT

If you have an offset of -46, -10, you can't probe the right edge of the bed, because you would get out of bed with a nozzle.
Use if you want to define an available probe plot

#if PROBE_SELECTED &&! IS_KINEMATIC
   #define PROBING_MARGIN_LEFT 11 // PROBING_MARGIN
   #define PROBING_MARGIN_RIGHT 47 // PROBING_MARGIN
   #define PROBING_MARGIN_FRONT 11 // PROBING_MARGIN
   #define PROBING_MARGIN_BACK 11 // PROBING_MARGIN
#endif

radek8, 2022-04-24:
> It is 26. Jan 35TFT and 24days ago for cinfig.ini
> 
> I updated 5 days back from: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

January 26 five days ago? Probably not...

bohomurcek, 2022-04-24:
I ment that I updated with files that were available 5 days ago on: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

radek8, 2022-04-24:
ok

bohomurcek, 2022-04-24:
Even if I program with M851 over pronterface(M851 X-49 Y-8), there is no response, X nd y offset always remain 0 like Eeprom
does not store that.

radek8, 2022-04-24:
Then it could be a mistake in Marlin.
Try updating Marlin

bohomurcek, 2022-04-27:
Thanks. I guess it is Marlin. Will wait till update and try again.

closed 2022-04-27 by bohomurcek

leoavilagg, 2023-04-03:
Perdon si mi mensaje esta fuera de contexto, pero este post lo visite para solucionar un problema con el m851 que no podia configurar en la eeprom, tengo un sensor sn04-n y con las siguientes lineas pude habilitarlo  para editar:

en Configuration.h

#define FIX_MOUNTED_PROBE
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define AUTO_BED_LEVELING_BILINEAR
Z_MIN_PROBE_ENDSTOP_INVERTING TRUE
#define Z_AFTER_HOMING 15