bigtreetech/SKR-2 #140: Faulty Mosfet?

AfroPsycho, 2022-04-22:
Hey
After I accidentally fried my first board
Someone supported me to buy second one ( it is too expensive to buy in Turkey )
This time while print is ongoing I smelled that something is burning 
Went to the room to checkout 
Print was done and printer was still working I thought it came out of a fan maybe as far as I know there is coal inside 
Anyway I shut the printer down opened the box where I keep my electronics 
Smell was definitely coming out of there but there was a cooling fan too
I powered on the printer again to check if fan is spinning 

Normally when SKR 2 powered on first red led lights up and after a second green led lights up with a click noise 

This time red led didn’t light up and green lighted up right away 
Bl touch tested itself screen have a light but no text 
I connect my rpi over tft port to use octoprint 
Octopribt cant connect to SKR 
it is not appearing in the pc 
I tried to power it on over usb but nothing has changed 
I can’t flash firmware either 

while inspecting the board I found this
[image]
What I am thinking is that mosfet probably died 
What else can be the problem 
Can I just swap that mosfet with my dead board

hcet14, 2022-04-22:
You may have a look here [https://github.com/bigtreetech/SKR-2/issues/102#issuecomment-1053588361](url)
In your case, I would unsolder the mosfet and disconnect everything. Power the board by USB and look if your tty appears on your rpi. Good luck!

AfroPsycho, 2022-04-23:
> You may have a look here [https://github.com/bigtreetech/SKR-2/issues/102#issuecomment-1053588361](url)
> In your case, I would unsolder the mosfet and disconnect everything. Power the board by USB and look if your tty appears on your rpi. Good luck!

You’re awesome thank you
Here is a question for you 
You said we need ~4,2a@24v
But my system is 12v so can I still use the one you suggested?

hcet14, 2022-04-23:
> You said we need ~4,2a@24v
> But my system is 12v so that means I need something around 8.4 amps which means more heat (?)

The IRFR5305PBF is more than suitable. I took that one for my SKR2 as a replacement.

AfroPsycho, 2022-04-23:
> > You said we need ~4,2a@24v
> > But my system is 12v so that means I need something around 8.4 amps which means more heat (?)
> 
> The IRFR5305PBF is more than suitable. I took that one for my SKR2 as a replacement.

Thanks a lot

closed 2022-04-23 by AfroPsycho