bigtreetech/SKR-2 #141: Dual Z port

MedSim9906, 2022-04-25:
Hi, 

I recently bought a SKR 2 from Amazon, got Klipper running and connected my 2nd z-axis to the E1 port and Changed my printer.cfg

But as soon as i want to Home z, Klipper Crashes due to No communication to The tmc 2209

Things i Tried: 

Things i Tried:

Changing the Driver (working x-stepper Driver)->Same issue

Connecting the new z_stepper motor to it -> Same issue

Replace The old z Stepper with the new one -> new one moves 

Wiring it in the parallel z Port and change config to one z Port with 1.6 Run current -> moving but the new one Skips pretty much every step 

Im totally Out of ideas....

Can anyone probably Help me Out on this?

Thanks a Lot

radek8, 2022-04-25:
The 1.6A current is quite high for some TMC2209. Poor cooling can overheat and lose steps.
Try to reduce the current

MedSim9906, 2022-04-25:
At first i tried it with 0.8A, because i forgot to change it and it skipped the steps. 
According to the Schematic the two ports are connected parallel. And this made me think i need to double up the Amps. 

I also checked the MOSFET to see if its REV A or B and there is nothing written on it.

MedSim9906, 2022-04-29:
After i swapped my skr 2 for a new, the problem is solved.

closed 2022-04-29 by MedSim9906