bigtreetech/SKR-2 #142: SKR-3 no printer attached

Gilgames1956, 2022-04-26:
Hi, attached the result of M122 command  in
[image]
 Pronterface:

radek8, 2022-04-26:
https://github.com/bigtreetech/SKR-2/issues/128

hcet14, 2022-05-03:
Have you connected MGND and PGND? [https://github.com/bigtreetech/SKR-2/issues/129](https://github.com/bigtreetech/SKR-2/issues/url)

radek8, 2022-05-04:
@hcet14  
Your URL links never work. When are you making a mistake?

radek8, 2022-05-05:
Don't click on the icon.
Just copy the page address and paste it on a new line via Ctrl + V

radek8, 2022-05-05:
You have to delete the word (url) between the parentheses before you insert the link there ....

hcet14, 2022-05-06:
> Don't click on the icon.
> Just copy the page address and paste it on a new line via Ctrl+V

I don't like the look, but thanks anyway.

> You have to delete the word (url) between the parentheses before you insert the link there ....

Doesn't work at all, but thanks anyway.

radek8, 2022-05-06:
Write the word you want to appear in square brackets and put the link itself in round brackets instead of the word URL

[my link](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues)

hcet14, 2022-05-07:
that's it

`["url"](url)`

I like the look.