bigtreetech/SKR-2 #144: Board won't start with 24v but starts with USB

landif, 2022-04-26:
I brought a new SKR2 rev B with 5x tmc2226.
I connected the board directly via USB and configured, installed new firmware, etc... After that, I replaced my old board with this SRK2 but the board refuses to start on direct power mode (24V). I even replaced the two 10A fuses with brand new ones in the hope that was a simple fuse issue but no luck... 
The board continues to start normally with a USB connection and jumper set to USB power but the TMCs won't start that way...
This is a hardware failure or am I missing something?

I created a vídeo showing the board's behavior.

[https://youtu.be/HyRP1Woh2oc](url)

Thanks

radek8, 2022-04-27:
If you checked fuse F2, there will be a problem with the 5V regulator.

[image]

[image]

radek8, 2022-04-27:
Can you also check fuse F1?

[image]

hcet14, 2022-04-27:
[https://github.com/bigtreetech/SKR-2/issues/137#issuecomment-1110907017](url)

radek8, 2022-04-27:
@landif  writes that he has revision B

hcet14, 2022-04-27:
> @landif writes that he has revision B

Please read! There is no difference between Rev A and B in my humble opinion.

radek8, 2022-04-27:
Revision A has used an inappropriate Mosfet, which will cause TMC drivers to burn out and other problems. Version B should already have the correct Mosfet type.
This board has been in constant trouble since its release.
I'm glad I'm not a "happy" owner :-)

hcet14, 2022-04-27:
> This board has been in constant trouble since its release.
> I'm glad I'm not a "happy" owner :-)

D'accord.

Please read [https://github.com/bigtreetech/SKR-2/issues/129](url)

...and especially [https://github.com/bigtreetech/SKR-2/issues/63#issuecomment-933762958](url) fitch22 explains the bug quite well!

landif, 2022-04-27:
I measured the F1 with power on vdd and nothing....
[image]

Probably this behavior isn't related with the problem found in rev A. The board has a HY1904 mosfet...

radek8, 2022-04-28:
Good. The fuse looks fine has 0 Ohm.

Turn on the power.
Remove the VDD jumper.
Measure the voltage at the VDD pin.

[image]

landif, 2022-04-28:
Measuring:
Power- (black wire) with vdd pin (top one) = no power
Power- with 5v (middle pin) = no power
Power- with USB (bottom pin) = no power
Power+ (red wire) with vdd = 24v
Powe+ with 5v = 24v
Power+ with USB = 12v

Immediately after power on the PSU i got 5v between power- and usb pin. But it goes away quickly and then goes to no power....

landif, 2022-04-28:
Thank You for your help!

I already filed a complaint within BTT's support system but no response so far...
I will change back my old board to have the printer working again while I try to solve the devolution of the board (and that can take some time due to the current BTT's support speed).
Probably my next printer will not have a BTT board!

hcet14, 2022-04-28:
Does D12 lights up, when you power up with a power supply?

radek8, 2022-04-28:
If the processor does not have power, there is no signal on the MOT POWER pin and LED D12 cannot light.

landif, 2022-04-28:
@hcet14, no lights... You can see this in the video on the fisrt comment.
I prefer to not try the @radek8 5v external power solution until BTT at least reply my ticket on the support system.

radek8, 2022-04-28:
BTT won't help you. They'll just tell you to deal with the dealer.

hcet14, 2022-04-28:
Then your board is missing VBB (Q2)! Have a look at the schematics.

[image]
You may use M42 to set the enable signal for the optocoupler. Check out if the pin is set or measure the output (PC13).

radek8, 2022-04-28:
VBB is 24V for fans and hotends

hcet14, 2022-04-28:
> The board continues to start normally with a USB connection

landif, 2022-04-28:
Q2 is on the board...
I think @radek8 is right... something is wrong with the power regulator (step down) when the power source is VDD 24v... 5v simpĺy can be measured on any part of the board.
When connected via USB, the 5v is there and goes directly to the processor, and the board boots!

> BTT won't help you. They'll just tell you to deal with the dealer.
They are the dealer (official store on aliexpress)

hcet14, 2022-04-28:
Well, if you think so. I measure.

hcet14, 2022-04-28:
Why don't you start with M42? D12 should light up!

landif, 2022-04-28:
this procedure is too advanced for me...

hcet14, 2022-04-28:
No, have you ever build (compiled) your own firmware for your SKR 2?

landif, 2022-04-28:
Yep... and I use Klipper... Marlin sucks

hcet14, 2022-04-28:
> Marlin sucks

Why do you think so?

hcet14, 2022-04-29:
> Kilpper

Very interesting approach for 3D printing, but I'm a beginner and have no clue about printing. I just worked me into Marlin and stay with it for a start. I even have a RPi, but used for other stuff.

Anyway your problem is easy to solve. Have a look at the schematics I posted above (or the original ones). Q2 enables VBB, You didn't set the pin which goes to Q2 in Klipper.

[image]

MOT POWER

landif, 2022-04-29:
It's not a firmware problem...
I flashed Marlin compiled from this repo and the board still has no response...

hcet14, 2022-04-30:
> Does D12 lights up, when you power up with a power supply?

radek8, 2022-05-01:
> @hcet14, no lights... You can see this in the video on the fisrt comment. I prefer to not try the @radek8 5v external power solution until BTT at least reply my ticket on the support system.

Isn't that enough for you to answer?

landif, 2022-05-03:
@hcet14 see the video in the first comment...

Well, one week and no response from BTT...
This board was the last product of this company, I ordered a new board from another vendor (a FLY-GEMINI plus tool head with CAN and accelerometer) and switch back to the old board meanwhile...
Thank you for your help!

BTT, shame on you...

hcet14, 2022-05-03:
> BTT, shame on you...

indeed...

Cool, this one? [https://github.com/Mellow-3D/Fly-Gemini-V1](url) 

If you are interested in selling you SKR 2, contact me... hcet@gmx.net

marcossfernandez, 2022-05-05:
I have 8 BTT SKR2 F429 bricked with correct 3.3v in the processor, regulators and fuses OK, all of them capable to flash firmware, **BUT lack of D12 led power light on (stay dim) and none of them can stablish USB comunications** after failure.

The USB esd diode protection apears to failed in all of them after the fail. It's frustating how all the boards ends this way. The cabling harness, pins and connectors are being double checked in all printers. Everything works OK after replace a fresh new board, but after some on and offs, test and calibractions, some updates and resets, these boards become a expensive brick.

It may be that reset lines entering the processor unfiltered, or voltages or data from the USB enter the processor through some part of the circuitry incorrectly under certain circumstances. Who knows! In order to be able to repair them, it would be essential to know which components fail, but above all, what really causes this failure.

We use these mainboards to drive our own commercial 3d printers and we  are obliged to replace and service in warranty to our customers. So the quality and agility from BTT sucks for a commercial product. 
We are looking for and alternative board or balance use a Duet3 (better hw) and RRF instead of Marlin.

radek8, 2022-05-05:
@marcossfernandez  Marco your problem will be related to what @hcet14 writes.
The SKV2 board has a design problem.
The board has implemented protection against incorrect connection of stepper motor controllers. Unfortunately, this is what causes the controls or the board to burn out.
Several earth potentials are separated on the dsece, which are switched via Mosfet transistors. Unfortunately, this leads to the generation of voltage peaks during switching on and off, which cause burnout.
The only protection is to remove these first and the replacement is a permanent connection.
It has been discussed several times in this repo.

marcossfernandez, 2022-05-05:
This problem hadn't  fixed on th rev. B boards? 

The new F429 rev B boards have the HY1904 mosfet that was supposed it address the issue on the rev A boards. 
The F407 rev B boards came with this new mosfet also. We have a rev A board that it has the oldest mosfet, but fixed soldering cables on bottom and uncomment the SNAFU part in the FW.

The MKS Robin V3.1 (eagle in some shops) have similar problems?  It seems to be the most convenient replacement and less tweak needed to replace SKR2 boards.

radek8, 2022-05-05:
https://github.com/bigtreetech/SKR-2/issues/132#issuecomment-1103440250

landif, 2022-05-05:
It's not only HY1904 mosfet that affects this board. Other design flaws related to power are causing other failures...
MKS Robin v3.1 can replace SK2. They are at the same level in terms of features.
BTT is not a good company to relay your commercial products on. Makerbase is slower in terms of software support (firmware updates and stuff). But if you choose to maintain your set of software to deploy in your commercial solution, Makerbase has far better hardware and post-sell support. 
As I said, that SKR2 (that BTT probably won't replace because they don't care about customers) was the last product that I acquire from this company. They do not even bother to check git's issues, reply in their own forums, read support tickets, or reply to buyers' messages in Aliexpress...