bigtreetech/SKR-2 #145: SKR 2 HE0 and HE1 not working.

eboston, 2022-04-26:
I have tried two different SKR 2 boards and on both of these board, I cannot get any power on the HE0 and HE1 connectors.  When I turn them on, I do see a voltage of 1.7v, but that is it.  If I connect the heater to the connector for the bed and turn it on, the heater works.  Both these board have the F429 processor.  Did the ports for these output change?

hcet14, 2022-04-27:
Does D12 lights up, when you power up with a power supply?

eboston, 2022-04-27:
The 5V LED lights up when I turn the power on to the board.  No LED lights when I turn on either HE0 or HE1.

hcet14, 2022-04-27:
Powerd by power supply? 12V or 24V, for curiosity's sake.

eboston, 2022-04-27:
24v.  Not going to fool around with it any more.  Found a SKR E3 Turbo that I can use instead.  Sending the SKR 2 boards back.

eboston, 2022-04-27:
Turns out the SKR E3 Turbo boards were not available, so I am still stuck with this problem.  :(

hcet14, 2022-04-27:
D12?

eboston, 2022-04-27:
D12 is the "power" light.

hcet14, 2022-04-27:
No.

Does D12 lights up,?

[image]

eboston, 2022-04-27:
I will check on that.  Right now I can't power the board up for a bit.

eboston, 2022-04-27:
D12 does NOT light up.

hcet14, 2022-04-27:
ooups Q2.

[https://github.com/bigtreetech/SKR-2/issues/140](url)

[https://github.com/bigtreetech/SKR-2/issues/102](url)

eboston, 2022-04-27:
If I understand what you are saying, there has been some issues with Q2.  Seems strange that two different boards acquired several months apart have the same issue.  Guess I will have to try a third board.

hcet14, 2022-04-27:
No!!!

Sorry, I'm tired, I'll get back tomorrow.

eboston, 2022-04-28:
Okay.  Think I got the problem fixed.  While I was waiting to find a solution for this, I decided to start testing the axis movement.  I hadn't installed any of the drivers as I like to make sure everything else is working first.  Once I installed drivers for the Z axis (Z and Z1), D12 lit up and the heater started working.  I have drivers in X, Z and E1 (E1 is driving the second stepper on the Z axis).

So, it seems you have to have drivers installed to get the heaters to work.  This might be a Klipper issue.  I'm not sure.  So, if you are using Klipper, make sure you have the drivers installed.

eboston, 2022-04-28:
One other thing I added to the Klipper config was:

[output_pin motor_power]
pin: PC13
value: 1

Not sure if this had any effect on the problem though.  It disables the anti-reverse protection on the stepper drivers.  It is only used by Marlin at this time.