bigtreetech/SKR-2 #147: SKR 2 F429 do not update FW from SD card

davbcr, 2022-05-04:
I started compiling [Marlin 2.0.9.3 MSU](https://github.com/PierreMasselot1/Marlin/tree/bf95c53990779534b821ca1b06bc9c16b2ad3916) because of the MSU project i would realize and i need to tweak my FW a bit before getting it ready to print.
I start having spontaneous reboot of the card untill i commented out : 
#define USB_FLASH_DRIVE_SUPPORT
#define USE_OTG_USB_HOST
#define NO_SD_DETECT
#define MULTI_VOLUME
and flash it but after that moment i cannot get to update the FW anymore via SD card nor USB.
Via USB VS code recognize the COM3 where the SKR2 is connected but the upload fails anyway.
I'm trying now to use all the SD card i have reformatting to FAT32 (not wuick formatting) and see what happens even plugging into TFT70 connected to the board.
Attached the config files i used (where the #defines listed above are being uncommented again.
[Marlin.zip](https://github.com/bigtreetech/SKR-2/files/8618401/Marlin.zip)
Is there a way to hard reset the board as if it has not any FW on it and waits for the first flash?

hcet14, 2022-05-04:
Your bootloader is probably missing. Have a look here [https://github.com/bigtreetech/SKR-2/issues/137](url)
You may ask BTT for the bootloader. service001@biqu3d.com

davbcr, 2022-05-04:
Oops..... i try to write them but about procedure to flash it?
I found this https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/115#issuecomment-653814967 do you think procedure can be the same?

hcet14, 2022-05-04:
> By the way board is alive, how can it loose the bootloader but work good for movement or PID or extruding?

I don't know.

You may start reading here [https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader](url)

davbcr, 2022-05-04:
for sure i'll keep and read it very carefully, but it appears that was the SD card not liked to the board, i put inside a very old 128MB one and it worked, so bootloader is not my problem for sure.

closed 2022-05-04 by davbcr