bigtreetech/SKR-2 #152: Mellow ESP32 WIFI on SKR2.0

sheski3, 2022-05-12:
Hi,  I have been very unsuccessful on adding either an ESP8266 S-12 or a 3dMellow ESP32 module to my SKR2.0 vB boards.  What is needed to prepare the wifi modules for use?  If I just plug them into the SKR2, they are not recognized.  There seems to be no guide on the internet that explains how to do this.  
Could someone please post a link on these instructions (if any) that my security system will allow?  3dMellow's website is blocked by all three of my internet security systems, and all the information on the ESP8266-S12 does not apply to a SKR2 board.

Thanks,

Jerry

jaysuk, 2022-05-12:
What firmware are you using?

sheski3, 2022-05-12:
I am using the firmware that came on the chip.  Bigtree sells the esp8266 on their website for the SKR2, but provides no installation instructions.  I am using RepRap firmware on the board and can connect to it with USB.  I compiles the firmware using instructions on the web.
I bought the ESP32 chip from 3dMellow as well.  I can't even find the pinout for that chip as I can't access their website.
Thanks for responding.

sinole1987, 2022-05-19:
> I am using the firmware that came on the chip. Bigtree sells the esp8266 on their website for the SKR2, but provides no installation instructions. I am using RepRap firmware on the board and can connect to it with USB. I compiled the firmware using instructions on the web. I bought the ESP32 chip from 3dMellow as well. I can't even find the pinout for that chip as I can't access their website. Thanks for responding.

he means what is your printer firmware. marlin or reprap firmware. the 8266 and mellow esp32 chips come equipped with reprap Wifi, so they wont work with marlin. to setup for marlin you might need a chip encoder. see https://github.com/luc-github/ESP3D

sheski3, 2022-05-20:
As I said above, I am using RepRap firmware on the SKR2.  Am I to understand that the ESP8266 and the ESP32 are essentually plug and play? I have included all the software that is needed for the controller to access the wifi module.

sinole1987, 2022-05-20:
well yes then it should be very simple. QC on skr 2 was a bit bad , ultimately you might have a faulty board. sorry i cant eally help you with rrf stuff because i tired to install it on my printer but I have  stuff like pinda 2 that you can really configure easy on rrf. but as far as I know Andy shaw aka gloomyandy is involved with mellow stuff maybe you can find some info on his github. https://github.com/gloomyandy

sinole1987, 2022-05-20:
im sure you went through these before but just in case. https://teamgloomy.github.io/skr_2.0_connected_wifi.html

sinole1987, 2022-05-20:
dumb question! did you change the red jumpers from usb to wifi?

jaysuk, 2022-05-20:
The WiFi modules come with no firmware installed. You need to use M997 S1,
as detailed ok the wiki, to load any firmware on to the ESPs

sinole1987, 2022-05-20:
> The WiFi modules come with no firmware installed. You need to use M997 S1, as detailed ok the wiki, to load any firmware on to the ESPs
> […](#)

great tanks for update, I bought one of those esp 32s its on transit, this will help. i currently use 8266 it came with reprap installed

sinole1987, 2022-05-24:
> 

apparently you need to be connected to board via WIFI and connected to module to program it. did you find a tutorial on this?
this is all I could find : https://teamgloomy.github.io/stm32_esp32.html 
its not enough

jaysuk, 2022-10-11:
It's pin compatible with the SKR2.
Neither BTT ESP8266 or the Mellow ESP32 come with the relevant firmware
installed for working with RRF.
You have to flash the ESP using RRF itself.
Details for board.txt, appropriate WiFi firmware and gcode to use are here
https://teamgloomy.github.io/skr_2.0_connected_wifi.html