bigtreetech/SKR-2 #154: Homing issue

AfroPsycho, 2022-05-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Recently my motherboard is fried (BTT SKR 2.0 F429) while printing after receiving the new board (BTT SKR 2.0 **F407** ) I installed it today and it shows a tmc connection error 
I found out that two of my stepper driver (BTT TMC2209) is reporting all low 
I have tried to change stepper driver around and confirmed that two stepper drivers are not working I have unplugged them and tried to home all axis 
Z lowered a bit as usual X and Y make buzzing sound moved just a little bit and printer halted
I tried it again and same result
Than I tried to home the axis individually Z is fine but x and y is broken
Then I plugged x stepper motor to the z plug on motherboard tried to home the z axis (x motor) and it moved fine 
Then plugged the z motor to the x plug tried to home it made buzzing sound moved only little bit (possibly because the vibration) and printer halted

So something with marlin or mainboard is making that buzzing sound and I don't know what that is

### Bug Timeline

after replacing the F429 mainboard with f407

### Version of Marlin Firmware

2.0.x

### Printer model

3D4E 4GenPro 2R

### Electronics

Upgraded to SKR2 with TMC2209

### Host Software

OctoPrint

### Additional information & file uploads
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8747298/Configuration.zip)

hcet14, 2022-05-21:
Did you connect MGND and PGND? If not, you may read here ["https://github.com/bigtreetech/SKR-2/issues/129"](https://github.com/bigtreetech/SKR-2/issues/129). I'm pretty sure it's no marlin bug. I would suggest to delete ["https://github.com/MarlinFirmware/Marlin/issues/24219"](https://github.com/MarlinFirmware/Marlin/issues/24219)

ellensp, 2022-05-22:
Buzzing is normally incorrectly wired stepper drivers. (or broke stepper motor wires)

Some printers have different steppers motors with different wiring on different axies. So switching the wires on the back of the steppers is not a good idea as the cables are axis specific. All stepper plugs on the motherboard are the same so swapping them at the motherboard is a good test.  (and never swap the wires while powered up, that is nearly always fatal for the stepper driver) 

TMC error can be caused by bad stepper drivers. or incorrect set jumpers, or badly configured firmware. (the latter being slave address related and not in this case) 
Recently there have been reports of badly designed tmc2209's that when butted head to tail (as is very common) they short to each other electrically. If having weird issues / dying stepper drivers you cannot explain  Try some kapton tape between stepper drivers.

Homing... tmc2209's have a diag pin which is electrically connected to the endstop pins, by way of a jumper
If you using any endstops other then so called sensor-less homing  you must remove the diag pin jumpers under the stepper driver. (ie all the red square)
[image]

hcet14, 2022-05-24:
> Some printers have different steppers motors with different wiring on different axies......Recently there have been reports of badly designed tmc2209's that when butted head to tail (as is very common) they short to each other electrically.

Ineresting, I wasn't aware of this.

AfroPsycho, 2022-05-24:
> Did you connect MGND and PGND? If not, you may read here ["https://github.com/bigtreetech/SKR-2/issues/129"](https://github.com/bigtreetech/SKR-2/issues/129). I'm pretty sure it's no marlin bug. I [...]

Thank you for the suggestion but I don't want to modify a brand new board 
it is hard for me to purchase a new one and also I am afraid of broking it
I am talking to BIQU support at the same time but no progress at the moment

AfroPsycho, 2022-05-24:
> Buzzing is normally incorrectly wired stepper drivers. (or broken stepper motor wires)
> 
> Some printers have different steppers motors with different wiring on different axies. So switching the wires on the back of the stepper motors is not a good idea as the cables are axis specific. All stepper plugs on the motherboard are the same so swapping them at the motherboard is a good test. (and never swap the wires while powered up, that is nearly always fatal for the stepper driver)
> 
> TMC error can be caused by bad stepper drivers. or incorrect set jumpers, or badly configured firmware. (the latter being slave address related and not in this case) Recently there have been reports of badly designed tmc2209's that when butted head to tail (as is very common) they short to each other electrically. If having weird issues / dying stepper drivers you cannot explain Try some kapton tape between stepper drivers.
> 
> Homing... tmc2209's have a diag pin which is electrically connected to the endstop pins, by way of a jumper If you using any endstops other then so called sensor-less homing you must remove the diag pin jumpers under the stepper driver. (ie all the red square) [image]

There is a point you missed 
This is exactly same setup that I was using before 
it it was working with no problems at all

hcet14, 2022-05-24:
> This is exactly same setup that I was using before
> it it was working with no problems at all

Do you have some knowledge in electronic design? Imagine you "steal" (even for a very short moment, I'm talking about µs) somehow the electrical ground under power. What will happen?

I compared the schematics of the new SKR 3 ["https://github.com/bigtreetech/SKR-3"](https://github.com/bigtreetech/SKR-3) with our crap today and I like it. It seems BTT has learned from the mistakes they did with our SKR 2s. Also the whole documentation (the one they provide to us) of that board is much better. They even designed a push button in to set BOOT0 to high. Nice.

AfroPsycho, 2022-05-24:
> > This is exactly same setup that I was using before
> > it it was working with no problems at all
> 
> Do you have some knowledge in electronic design? Imagine you disconnect (even for a very short moment, I'm talking about µs) somehow one of the three electrical grounds (MGND, PGND, GND) under power. What will happen?
> 
> I compared the schematics of the new SKR 3 ["https://github.com/bigtreetech/SKR-3"](https://github.com/bigtreetech/SKR-3) with our crap today and I like it. It seems BTT has learned from the mistakes they did with our SKR 2s. The connection of MGND and PGND by a mosfet is gone or better the whole MGND disappeared. Also the whole documentation (the one they provide to us) of that board is much better (only the ground planes are still missing). They even designed a push button in to set BOOT0 to high. Nice.

I just now basics about electric & electronics 
What will happen

also I was thinking of trading my “brand new” skr 2 with skr3 ez 
What is your opinion 
Should I do it?

hcet14, 2022-05-24:
> What will happen

Well, it is all explained by VanessaE ["https://github.com/bigtreetech/SKR-2/issues/63#issuecomment-933916535"](https://github.com/bigtreetech/SKR-2/issues/63#issuecomment-933916535) I would add that it will also cause terrible transitions (big spikes) which will affect the digital I/Os everywhere on the board. I'm talking about destruction or at least severe damage of the ICs and so if not broken funny behavior. Just read here in the repository, there are so many issues which nobody can explain. I think these are caused to what I called 'conceptional design bug'.

> also I was thinking of trading my “brand new” skr 2 with skr3 ez

I like the layout of the SKR 3 more, but that is a question of taste :wink:
Better get a board, where you get support from the community. SKR 3 is way too fresh.

AfroPsycho, 2022-05-24:
> > What will happen
> 
> Well, it is all explained by VanessaE ["https://github.com/bigtreetech/SKR-2/issues/63#issuecomment-933916535"](https://github.com/bigtreetech/SKR-2/issues/63#issuecomment-933916535) I would add that it will also cause terrible transitions (big spikes) which will affect the digital I/Os everywhere on the board. I'm talking about destruction or at least severe damage of the ICs and so if not broken funny behavior. Just read here in the repository, there are so many issues which nobody can explain. I think these are caused to what I called 'conceptional design bug'.
> 
> > also I was thinking of trading my “brand new” skr 2 with skr3 ez
> 
> I like the layout of the SKR 3 more, but that is a question of taste 😉 Better get a board, where you get support from the community. SKR 3 is way too fresh.

Gonna read that

I don’t really want to purchase another skr2 board 
But I don’t see much of alternatives