bigtreetech/SKR-2 #158: Auto build Errors

chrisljames90, 2022-05-30:
Trying to compile my firmware and keep coming across some errors been trying to track them all down and im now down from 7 errors from the 30 I originally had.  1 and 2 are not preventing me from building.  but 3-7 are the ones making the build fail.  Should i change the names as it's recommending or leave it like it is and keep tracking?

1) Confguration.h - "define" does not name a type [ln 1445, Col 1]
2) serial_base.h (Marlin/src/core) - "serial_index_t" has not been declared [Ln 116/120/124/127, Col 17/12/26/20]
3) xpt2046.h 35,4 - "touch_miso_pin' is not defined.
4) xpt2046.h 72,94 - "touch_cs_pin' was not declared in this scope; did you mean "Touch_int_pin"
5) xpt2046.h 73, 48 - "touch_cs_pin" was not declared in this scope.
6) xpt2046.cpp 41,13 - "touch_cs_pin" was not declared in this scope; did you mean "touch_int_pin"
7) xpt2046.cpp 48,75 - "touch_sck_pin" was not declared in this scope; did you mean "touch_int_pin"

chrisljames90, 2022-05-31:
> Your pins_BTT_SKR_V2_0_common.h is wrong. You may have a look here
> Mint that I use a different display!

I got the firmware to build something was uncommented that should have been
commented.  Not I’m having problems getting the firmware to flash

hcet14, 2022-05-31:
> Not I’m having problems getting the firmware to flash

Well, probably your SDcard, but you may search/read here :wink:

hcet14, 2022-06-01:
> be nice if it was setup to flash from a usb thumb drive

Sorry, ["https://github.com/MarlinFirmware/Marlin/issues/24040#issuecomment-1100450573"](https://github.com/MarlinFirmware/Marlin/issues/24040#issuecomment-1100450573)

The size of your SDcard doesn't matter as long as Marlin fits. I also had some trouble to find a card SKR 2 would accept.