bigtreetech/SKR-2 #159: SKR2  heated bed causing printing issues

chiron5498, 2022-06-04:
i have an skr 2 with tft 35v3 installed on my chiron  with 2209s  and marlin 2x, everything works  except when you print with the heated bed ,the layer lines in the print are all unaligned, with the bed cold  the walls are perfectly smooth, tried bang bang and its no different. Have run the pid several times to no avail , have tried the latest marlin , new heated bed new psu, nothing fixes it.  never had issues with my trigorilla board , tried dropping bed power in marlin to 2/3

hcet14, 2022-06-04:
> The heated bed in the chiron is run by its own 24v psu , board is on its own 24v psu

Well, not exactly. This is no SKR 2 issue!
Your trigorilla board uses two psus. The SKR 2 uses one big psu. Have a look at the schematics.
[image]
[image]
You could connect your heatbed psu to F4 and remove the fuse.

BetulSBoyali, 2022-06-04:
Hey, it is because the power supply is not enough. Use 500watt instead. I had the same issue with another board, I tried everything, pid, bigbang.. nothing worked except replacing power suplly to higher watt.

hcet14, 2022-06-04:
> Hey, it is because the power supply is not enough. 

Correct, I agree.

> Use 500watt instead. 

I don't agree. A common printer with 4 steppers, one extruder, hotbed not bigger than 24x24cm, mainboard, and a common display should work with 350W without problems. I think this is a good article about power consumption ["https://clevercreations.org/how-much-electricity-3d-printer-power-usage"](https://clevercreations.org/how-much-electricity-3d-printer-power-usage)
When I bought a new hotbed (22x22cm), I did some tests.
[image]
The current wasn't bigger, when the bed started heating (I was expecting a much higher current).

chiron5498, 2022-06-05:
Well its an skr 2 not a trigorilla board and the chiron has 2 seperate
psu's one fot the board and the big one fir the bed and its 1000w so
shouldnt be a problem

BetulSBoyali, 2022-06-05:
Okay, I also changed the heat bed to aluminium one with the new power supply, it was a cheap thin heatbed before, which i assume to consume high power to heat up. Maybe isolation may work. I also read about signal noise between Bltouch and heatbed while heating up that can cause this kind of wobbling effect on the print. I don't know if that will work but i wrote a gcode on cura that may have stopped it.. That is M420 Z0 ; ... try if you have bltouch, it may work.

hcet14, 2022-06-05:
@BetulSBoyali 

How did you fix your issue? ["https://github.com/bigtreetech/SKR-2/issues/137#issuecomment-1110351643"](https://github.com/bigtreetech/SKR-2/issues/137#issuecomment-1110351643)

@chiron5498 

How many psus are you using now with your chiron and SKR2?

chiron5498, 2022-06-05:
Still 2 psu's

chiron5498, 2022-06-05:
That command is disabling the z fade height, which i already tried
disabling and didnt work thx

BetulSBoyali, 2022-06-05:
I couldnt fix it unfortunately.. I had a spare one, I am using that one.. I dont know what to do with it.

hcet14, 2022-06-06:
@chiron5498

> Still 2 psu's

What? How did you connect that? A little block diagram may help :wink:

@BetulSBoyali

Please pm me (hcet@gmx.net)

hcet14, 2022-06-07:
@chiron5498

[image]

You may try it like this

[image]

chiron5498, 2022-06-08:
I forgot to draw the external mosfet in there for the bed

chiron5498, 2022-06-08:
Well the part i had bed logic beside is the mosfet, the bed draws 21A , the
psu is 40a so i guess thats enough, i think they put it on a seperate psu
because the big bed causes to much electrical interference.kinda need the
mosfet dont i

hcet14, 2022-06-08:
> Well the part i had bed logic beside is the mosfet, the bed draws 21A , 

How big is the bed? 21A sounds a little high. Did you measure that?

> the psu is 40a so i guess thats enough, 

Oh yes 

Q2 is more than capable to handle the 21A, but I would measure the temperature of Q2 with your finger when it starts heating. There were some reports of blown Q2. I had to replace mine [""https://github.com/bigtreetech/SKR-2/issues/102](https://github.com/bigtreetech/SKR-2/issues/102)

chiron5498, 2022-06-09:
Bed is 410 x 410