bigtreetech/SKR-2 #163: SKR-2 RepRap ESP-07S, ESP-12S modules not connecting via USB

dsplendido, 2022-07-27:
I've got reprap firmware loaded on an SKR-2 and connect to it via YAT no problem. When I try to upload wifi firmware to the ESP module(s), I either get an failure to connect error at different baud rates (ESP-12S) or there is no com port whatsoever (ESP-07S).

fern1988, 2022-08-23:
Same issue.. iv now got the esp32u wifi module still same issue with yat. I can connect with putty get as far as uploading firmware but doesn't like ssid and goes idle when I start the module up. Iv gone tru 2 skr2 boards and I have the new mantra mp8 I don't even want to think about setting up with this board been such a turd. Not happy with email response from btt so I'm now here to vent :p you have any luck?

dsplendido, 2022-08-23:
Not enough luck to even continue trying. I've also got a couple of SKR2 boards and an Octopus. Sorry to hear you have the same. BTT was asking for the order # which I have, from MicroCenter, but I don't see what that has to do with the issue. Is it a known issue? I don't really want to send the board back, get a new one and go through the same trouble with probably the same result. Not fun.

dsplendido, 2022-08-23:
I've moved on to a couple of FLY CDY V3 and so far, so good. Got it up and running with no issues. Good support on discord.

jaysuk, 2022-08-24:
Join us on discord for support https://discord.gg/uS97Qs7

fern-cueplay, 2022-08-24:
> Join us on discord for support https://discord.gg/uS97Qs7
> 
Next port of call cheers Jay

fern-cueplay, 2022-08-24:
> I've moved on to a couple of FLY CDY V3 and so far, so good. Got it up and running with no issues. Good support on discord.
> 

Goner have another bash with this skr2 then might attempt another build with mantra and new cb1 see how guide fair for that with klipper.

dsplendido, 2022-08-24:
Ping me if you wouldn't mind if you come across a helpful tidbit on discord, cause it's just painful staring at those unused boards.

jaysuk, 2022-10-11:
Come join us on discord for help https://discord.gg/uS97Qs7