bigtreetech/SKR-2 #165: Exp1/exp2 logic level tolerance

mikerocklewitz, 2022-09-09:
Are the pins in exp1/exp2 5 volt friendly or 3.3v only?

radek8, 2022-09-09:
Look in the processor manual...

[image]

[image]
[image]
[image]

[stm32f405og.pdf](https://github.com/bigtreetech/SKR-2/files/9538375/stm32f405og.pdf)

mikerocklewitz, 2022-09-10:
> Look in the processor manual...
> 
> [image]
> 
> [image]
> [image]
> [image]
> 
> 
> 
> [stm32f405og.pdf](https://github.com/bigtreetech/SKR-2/files/9538375/stm32f405og.pdf)
> 

Brilliant.  Ill have to pull the f429 datasheet.  Never realized there was 5v tolerance on stm32 - i always thought it was 3.3v for everything and made a bad assumption bigtreetech added something in line somewhere to make it tolerant and never bothered to look at the datasheet -  you sent me on the right track for how to figure this out.

Again ‐ thank you so much that answers everything for me.

closed 2022-09-10 by mikerocklewitz