bigtreetech/SKR-2 #168: SKR-2 - reset button from the mainboard required to flash new versions of Marlin

Hasoth, 2022-09-12:
Hi,

During my board upgrade I've uploaded many version of the marlin 2.1. One of the upload ended up on freeze after restart and 30 mins of no response from the box afterh which I've decided to do hard restart.

Since that time I could not flash marlin firmware from SD card normally, every time I need to click small restart button on the board and then it loads new firmware from the SD card.

Any idea if I can bring it back to normal?

Hasoth, 2022-09-16:
Is anybody still looking in folder "issues" for SKR2? ;)

hcet14, 2022-10-23:
> Is anybody still looking in folder "issues" for SKR2? ;)

I don't think so ;)

> That is the same moment I've enabled "#define SDSUPPORT"

Disable "#define SDSUPPORT". I attached my config. All made entries are marked by 'hcet14'. Compiled with VS code and Auto Build Marlin and works with this setup [https://github.com/MarlinFirmware/Marlin/issues/24372#issuecomment-1215679054](https://github.com/MarlinFirmware/Marlin/issues/24372#issuecomment-1215679054) 

Good luck.
[marlin_config.zip](https://github.com/bigtreetech/SKR-2/files/9846408/marlin_config.zip)

Hasoth, 2022-10-23:
Hi,

first of all thanks a lot for replying!

I don't think I want to go for the solution, because if I won't use SDSUPPORT I will also not be able to use "USB_FLASH_DRIVE_SUPPORT" which is inside all of this part of the configuration_adv.h . This is really unfortunate but it looks like there's no good way to fix possibility to use auto-update from SD card an other stuff. I've tried some other options that don't work too (like #define MENU_ADDAUTOSTART)

There are even more bugs and I don't know if Marlin or the board... no matter what config, if I have USB and SD card enabled, I can use SD card only if USB is plugged in. Untill USB is not plugged in, the printer does not see the SD card at all :D

Well since I won't update the marlin so often anymore I can live with it, but it's sad there's no way to figure it out.

hcet14, 2022-10-24:
Hi Hasoth,

maybe I don't understand your issues in the right way (might be language problems from my side)?

Please list your issues (or what you wish to have) in a list, like for example

1: Use onboard USB-A connector
2: Not resetting the board when I turn it on
and so on...

Maybe I'm able to tell you, what you could have...

I agree, the board has many hardware bugs, but if you are a good man using a soldering iron, no problem at all. The most important bug is this ['https://github.com/bigtreetech/SKR-2/issues/129'](https://github.com/bigtreetech/SKR-2/issues/129)

Please also read all the links mentioned in this thread!

The SKR2 runs good with marlin!

Kind regards, hcet14

Hasoth, 2022-10-26:
Hi,

I'd like to:
* Use SD card for printing
* Use USB pendrive for printing
* when I load new firmware for SD card, I'd like printer to install new firmware as before and restart automatically (currently I need to unscrew the box, take out the poweradapter and physically click small button on the mainboard)