bigtreetech/SKR-2 #169: shorted PB5 & VBB need help

discip, 2022-10-04:
So, as mentioned in the title, I managed to short out one of the WPM fan outputs.

Now I need to find out what got corrupted.

Is anyone willing to help me diagnose if the MCU is fried or if I just need to replace some resistors, capacitors or something?

Additional information:
I tried powering the board from both DCIN and USB.
In both cases the connected display shows **`no printer attached`**.

Inserting an SD card with a **`frimware.bin`** leaves the mentioned file untouched.

The green LED next to the fan output that I shorted out is on because the corresponding transistor is also fried.

This is what it looks like when powered up:

[image]

thanks in advance
kind regards

discip, 2022-10-04:
Thank you for the quick reply!
F2 is ok!
Other than that I have no idea where to locate Q2 & Q10?😅

radek8, 2022-10-04:
If you connect the board to the computer via USB, does it communicate with the computer?

discip, 2022-10-04:
No it does not!

radek8, 2022-10-05:
Check fuse F1 (T300-4A)
If it is OK and there is 3.3V power on the wifi connector
So the processor will be burnt.

radek8, 2022-10-05:
2 red LEDs are on, so the 3.3V power supply will be fine.
Looks like a faulty MCU :-(

discip, 2022-10-05:
Thanks for the effort.

Have a nice day. 🙂

closed 2022-10-05 by discip