bigtreetech/SKR-2 #175: Schematic and board layout

roontoon, 2023-03-13:
I have found the schematic for the SRK 2 but I also need to be able to locate the parts on the board. Do you have a drawing that includes the component locations on the SKR 2. Thanks

roontoon, 2023-05-23:
> Open the pdf file from this repository in acrobat reader.
> and you have the option to search for individual pieces that are highlighted on the board
> For example, U3 is here:
>
>  <https://user-images.githubusercontent.com/46979052/240352199-b6dcdd25-9b8b-4833-88c8-b6859f502427.png>

Thank you for your response. That is a diagram of the part placement. I need a schematic so I can trace a fault. Is it not OpenSource?

radek8, 2023-05-23:
the schematic is here too...

[image]