bigtreetech/SKR-2 #176: TMC2209 v3.0 UART communication BTT SKR2

largocreatura, 2023-06-01:
Hello, I have a BTT SKR2 board and Fysetc TMC2209 v3.0 drivers. The board has Marlin firmware loaded  (I've tested on several versions, and same error) to work with the rev b model board and F429 chipset. I have set the jumper to UART mode in the skr2 board near driver socket. The board is powered with 24V and I set the pin to VDD mode. When I connect from Pronterface and send M122, it returns the following messages: "device registers: bad response!" and also "Error: All LOW." The LCD displays the message "TMC Connection Error." The thing is, I can move the stepper from Pronterface, so it seems that the driver is not damaged, but I can't establish the UART connection. Do you know what could be happening? In other SKR models, I have seen that they connect the Rx and Tx pins with a jumper for UART communication, but I haven't found any information about this for the SKR2 board. I hope you can help me identify the problem. Thank you!

radek8, 2023-06-09:
You have to cut the jumper for the UART mod
I think he's here

[image]

radek8, 2023-06-09:
The problem may be in connecting the Uartt inputs to other pins at Fysetc and BTT.
https://theres-probably-a-better-way-to-do-it.readthedocs.io/en/latest/pages/electronics/skr_1.3_and_fysetc2209v3.html

radek8, 2023-06-09:
Versions FISTEC 3.1 and 4.0 have a different connection
Apparently a good company :-/

[image]

[image]